carla-simulator/carla #2849: Custom background vehicle model in Carla

miracelplus, 2020-05-12:
Hello,

Thanks for making a great simulator. I am doing a project in which I need to **customize all vehicles' dynamic models** in the Carla environment (for example, the vehicle will get the ground truth information and decide to change the lane or change the acceleration).

In order to get information about all the vehicles, I use the following codes:
```
waypoint_list = [world.get_map().get_waypoint(v.get_location(), project_to_road=True) for v in vehicles_list]
```

As I have about 200 vehicles in the Town04 map, I noticed that the line above cost me about 5 seconds each time, which can not meet the real-time requirements.

I wonder if there are other solutions to let me **customize the control model** of the vehicles using the **provided PythonAPI**, or I may need to modify the source code of Carla to reach my purpose?

Thanks!

germanros1987, 2020-06-26:
@miracelplus you just want to get the location of all the vehicles for a given time instant? @glopezdiest could you provide a more effective way to  do this?

glopezdiest, 2020-06-26:
Hey @miracelplus. Could you provide a little more detail to your process? What are you gonna use the waypoint_list for? Isn't the location of the vehicles good enough? From what you've showed, the only recommendation i can tell you is, don't get the map every time, store it in a variable once and call it to get the waypoints.

miracelplus, 2020-06-29:
Hello,

Thanks for your reply! I’m working on customizing my own vehicle model (car following and lane change). The detailed usage of my application is like this:

1.For every second, I need to get the information for each vehicle
2.Then, I decide the next-second acceleration for this vehicle. 
3.I use waypoint for each vehicle and get the acceleration from my own decision part in order to keep the vehicle on the lane. 
4.Calculate the next waypoint the vehicle should achieve given specific acceleration.
5.Teleport the vehicle to the new waypoint

That’s how I finish my customized car-following model. I know this may be a dumb usage of provided PythonAPI. Therefore, I wonder if there’s some more efficient ways that I can fulfill my goal?

Thanks!

glopezdiest, 2020-06-29:
I see. I'm still a bit confused though.

From what I understand, you only care about the acceleration of the vehicle, which you are using to move (teleport) the vehicle. Are you using this specific approach instead of using a controller for some reason?

Why are you using the waypoints instead of the location, because they are in the middle of the road so you only care about going forward?

You seem to only care about using waypoints but you are getting them from the vehicle location on each frame (the command you showed at the beginning of this issue). So how are you calculating this "next waypoint"? You can just use the *carla.Waypoint.next()* function, which should remove the use of that location-to-waypoint-loop, speeding up the process.

This any of this makes sense? I may have just completely missunderstood what you are trying to achieve

miracelplus, 2020-06-29:
Hello,

I’m using this approach instead of a controller is because that some of my customizing model need all the vehicles’ information around one specific vehicle. Therefore, I choose to use the PythonAPI to get all the vehicle’s information and individually control each of them. I wonder if the controller method can get the information of surrounding vehicles (like 5 or 6 vehicles around this vehicle).

Also, the usage of waypoint method is that I wish the vehicle can follow the road, for example, the vehicle is on a curve shaped road and I wish to control the “longitudinal” acceleration of it. Is there some more straightforward method to achieve this goal? I’m only using Carla for a short period of time so I may miss some more useful approaches.

The waypoint.next method is very helpful to my usage. Deleting the location-to-waypoint method accelerates the running speed.

Thanks for your reply, that means a lot to me.

glopezdiest, 2020-06-30:
> Is there some more straightforward method to achieve this goal?

No, I think you are doing the correct approach here, but I've never though about it so I was confused. 

> The waypoint.next method is very helpful to my usage.

Good to hear :+1:. I'm leaving this issue open in case your solution isn't final, but please, remember to close it if you find that it's resolved

miracelplus, 2020-06-30:
Thanks a lot!

closed 2020-06-30 by miracelplus