carla-simulator/carla #2850: Can't install compiled eggs

tom-doerr, 2020-05-13:
I compiled the Python egg packages as described in the documentation (https://carla.readthedocs.io/en/latest/build_system/).
However when I try to install them I get the following error message:
```
Processing carla-0.9.7-py3.6-linux-x86_64.egg
Traceback (most recent call last):
  File "/home/user1/.pyenv/versions/3.6.9/bin/easy_install", line 11, in <module>
    sys.exit(main())
  File "/home/user1/.pyenv/versions/3.6.9/lib/python3.6/site-packages/setuptools/command/easy_install.py", line 2316, in main
    **kw
  File "/home/user1/.pyenv/versions/3.6.9/lib/python3.6/site-packages/setuptools/__init__.py", line 143, in setup
    return distutils.core.setup(**attrs)
  File "/home/user1/.pyenv/versions/3.6.9/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/user1/.pyenv/versions/3.6.9/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/user1/.pyenv/versions/3.6.9/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/user1/.pyenv/versions/3.6.9/lib/python3.6/site-packages/setuptools/command/easy_install.py", line 418, in run
    self.easy_install(spec, not self.no_deps)
  File "/home/user1/.pyenv/versions/3.6.9/lib/python3.6/site-packages/setuptools/command/easy_install.py", line 660, in easy_install
    return self.install_item(None, spec, tmpdir, deps, True)
  File "/home/user1/.pyenv/versions/3.6.9/lib/python3.6/site-packages/setuptools/command/easy_install.py", line 705, in install_item
    dists = self.install_eggs(spec, download, tmpdir)
  File "/home/user1/.pyenv/versions/3.6.9/lib/python3.6/site-packages/setuptools/command/easy_install.py", line 851, in install_eggs
    return [self.install_egg(dist_filename, tmpdir)]
  File "/home/user1/.pyenv/versions/3.6.9/lib/python3.6/site-packages/setuptools/command/easy_install.py", line 909, in install_egg
    dist = self.egg_distribution(egg_path)
  File "/home/user1/.pyenv/versions/3.6.9/lib/python3.6/site-packages/setuptools/command/easy_install.py", line 897, in egg_distribution
    metadata = EggMetadata(zipimport.zipimporter(egg_path))
zipimport.ZipImportError: not a Zip file: 'carla-0.9.7-py3.6-linux-x86_64.egg'
The command '/bin/sh -c bash -ic 'easy_install --user --no-deps carla-0.9.7-py3.6-linux-x86_64.egg'' returned a non-zero code: 1
```

Any ideas what the issue could be?

germanros1987, 2020-06-26:
@sergi-e could you please help with this and perhaps update the documentation accordingly?

Jaskaran-Singh-Sodhi, 2020-07-29:
I had a similar error but I'm not sure what to do

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 360, in run
    prefix=options.prefix_path,
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 784, in install
    **kwargs
  File "/usr/lib/python2.7/dist-packages/pip/req/req_install.py", line 851, in install
    self.move_wheel_files(self.source_dir, root=root, prefix=prefix)
  File "/usr/lib/python2.7/dist-packages/pip/req/req_install.py", line 1064, in move_wheel_files
    isolated=self.isolated,
  File "/usr/lib/python2.7/dist-packages/pip/wheel.py", line 478, in move_wheel_files
    generated.extend(maker.make(spec))
  File "/usr/share/python-wheels/distlib-0.2.6-py2.py3-none-any.whl/distlib/scripts.py", line 403, in make
    self._make_script(entry, filenames, options=options)
  File "/usr/share/python-wheels/distlib-0.2.6-py2.py3-none-any.whl/distlib/scripts.py", line 307, in _make_script
    self._write_script(scriptnames, shebang, script, filenames, ext)
  File "/usr/share/python-wheels/distlib-0.2.6-py2.py3-none-any.whl/distlib/scripts.py", line 281, in _write_script
    self._fileop.write_binary_file(outname, script_bytes)
  File "/usr/share/python-wheels/distlib-0.2.6-py2.py3-none-any.whl/distlib/util.py", line 548, in write_binary_file
    with open(path, 'wb') as f:
IOError: [Errno 20] Not a directory: '/home/jaskaran/.local/bin/easy_install'

closed 2020-10-04 by stale[bot]