carla-simulator/carla #2851: Carla 0.9.8 Crash - VK_ERROR_DEVICE_LOST

Jack-Pinnington, 2020-05-13:
Hello,

I'm experiencing intermittent crashes.

Here's the console output:

./CarlaUE4.sh 
4.22.3-0+++UE4+Release-4.22 517 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 678] 
VulkanRHI::vkDeviceWaitIdle(Device) failed, VkResult=-4
 at /home/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp:1015 
 with error VK_ERROR_DEVICE_LOST
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=145408 LargeMemoryPoolOffset=276544 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

This is CARLA 0.9.8, Ubuntu 18.04.
My use-case is that I am replaying a .log file and saving sensor output to files.
Do you have any ideas what could be causing this error or what I could do to address it?
I've tried reducing the number of vehicles in my .log file from 150+ down to below 100 which helped but the issue is still present. 

Any help is appreciated.

Thanks,
Jack

Jack-Pinnington, 2020-05-13:
Since this might be relevant - Graphics Card is NVIDIA GTX 1060 running nvidia-driver-435.

Jack-Pinnington, 2020-05-13:
Okay - closing this issue for now, but here's what I've found.
Graphics Card is NVIDIA GTX 1060 6GB for future reference - Ubuntu 18.04, Carla 0.9.8.
My simulation included over 100 vehicles, replayed from a .log file, and six RGB sensors on my egoVehicle.

Issue seems to be caused by bad Malloc in the GPU (Vulkan error).
Tried several methods to use less memory in the GPU to prevent this, including:
- Attaching the six sensors two-at-a-time (ie 3 runs rather than 1), to limited success,
- Freeing up more Disk Space on my Hard Drive (from 85% capacity to 30% capacity),
- Reducing the number of image saving threads.

These all had some effect but some maps are just bigger than others (Town04 > Town01) which will require more GPU space and thus fewer vehicles in the .log file.
Had to do a combination of all three techniques + smaller simulation - vehicles down to around 40.

Hopefully this helps someone else when they run into this error.

Best,
Jack

closed 2020-05-13 by Jack-Pinnington

pengpingliang, 2020-05-25:
@Jack-Pinnington 
Could you tell me how do you resolved it ?
I met the same error