carla-simulator/carla #2853: Carla make launch error

AdelinaK, 2020-05-13:
I tried to build Carla, this is my frist time working with it and after the make launch command I have this error 

Error: Couldn't find parent type for 'CarlaLightSubsystem' named 'UWorldSubsystem' in current module or any other module parsed so far.
make: *** [Util/BuildTools/Windows.mk:20: CarlaUE4Editor] Error 5

closed 2020-05-18 by AdelinaK

RehabHsn, 2020-05-24:
have you found a solution?

sourabhmisal, 2020-06-30:
I am facing this issue in ubuntu 18.04,  when I run `docker build -t carla -f Carla.Dockerfile .`

`Parsing headers for CarlaUE4Editor
  Running UnrealHeaderTool "/home/ue4/carla/Unreal/CarlaUE4/CarlaUE4.uproject" "/home/ue4/carla/Unreal/CarlaUE4/Intermediate/Build/Linux/B4D820EA/CarlaUE4Editor/Development/CarlaUE4Editor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors
Error: Couldn't find parent type for 'CarlaLightSubsystem' named 'UWorldSubsystem' in current module or any other module parsed so far.
make[1]: *** [CarlaUE4Editor] Error 5
Makefile:215: recipe for target 'CarlaUE4Editor' failed
make[1]: Leaving directory '/home/ue4/carla/Unreal/CarlaUE4'
make: *** [CarlaUE4Editor] Error 2
Util/BuildTools/Linux.mk:79: recipe for target 'CarlaUE4Editor' failed
The command '/bin/sh -c cd /home/ue4 &&   if [ -z ${GIT_BRANCH+x} ]; then git clone --depth 1 https://github.com/carla-simulator/carla.git;   else git clone --depth 1 --branch $GIT_BRANCH https://github.com/carla-simulator/carla.git; fi &&   cd /home/ue4/carla &&   ./Update.sh &&   make CarlaUE4Editor &&   make PythonAPI &&   make build.utils &&   make package &&   rm -r /home/ue4/carla/Dist' returned a non-zero code: 2
`

Dennis174698, 2020-07-01:
> I tried to build Carla, this is my frist time working with it and after the make launch command I have this error
>
> Error: Couldn't find parent type for 'CarlaLightSubsystem' named 'UWorldSubsystem' [...]

have you solve this issue?

ajuton-ens, 2020-07-03:
I have the same problem....

AdelinaK, 2020-07-03:
Yes. You need a 4.24 Unreal version. Make sure the CarlaUE4 file is selected to run on this version if you have more than one installed

ajuton-ens, 2020-07-06:
I built the docker with 4.24 Unreal version and it worked. Thank you.

qhaas, 2020-09-08:
> Yes. You need a 4.24 Unreal version
Ran into this problem today when trying to build with 4.23.1, looks like `UWorldSubsystem` [was added in 4.24](https://www.ue4community.wiki/programming-subsystems-29v70qij).