carla-simulator/carla #2854: Traffic Signal made by RoadRunner is available?

MingooJ, 2020-05-14:
Hi, I'm using Carla for fun.

I am interested in road traffic and make some road.
But after 'road.fbx and road.xodr' import to carla well, no car stop to my traffic signals.

How can i stop the autopilot car at traffic light?

Thank you in advance.

Mkarami3, 2020-06-12:
Hi MingooJ,
Did you manage to solve this issue? I have the same problem. Thanks

MingooJ, 2020-06-15:
Hi Mkarami3,
Unfortunately, this problem still unsolved. but, i generate traffic light of UE4, it works. 
However, complex traffic can't create with it's traffic system. So, i still try solving it.
Thanks.

germanros1987, 2020-06-26:
@MingooJ please provide further information. Which version of CARLA are you using? OS? etc.

@XGodina could you help with this?

MingooJ, 2020-06-26:
Hi, @germanros1987 

I used CARLA 9.8 in Ubuntu18.04 and made map using RoadRunner.
I made a traffic light on the roadrunner, but it doesn't seem to work on the CARLA.
And my roadrunner was a trial version, so I can't do anything right now.

I wish it could be of help someone's help.

Thanks.

germanros1987, 2020-06-26:
@MingooJ you need to use 0.9.9 to be able to ingest traffic lights automatically from OpenDRIVE (generated by RoadRunner). Then, you need to cook the contents using the docker provided by CARLA. 

@bernatx @sergi-e could you please add more details so that @MingooJ can have a smooth journey?

sergi-e, 2020-06-30:
Hi, @MingooJ ! 
Indeed, the automatic ingestion of landmarks was added by CARLA 0.9.9. If you [download package 0.9.9](https://github.com/carla-simulator/carla/releases) and follow [this tutorial](https://carla.readthedocs.io/en/latest/tuto_A_add_map/#map-ingestion-in-a-carla-package) to add the map, you should be able to run your custom map in CARLA with traffic lights in it.

closed 2020-09-05 by stale[bot]