carla-simulator/carla #2855: Problem with Map.get_waypoint() in combination with OpenDRIVE spiral

alexk-92580, 2020-05-14:
Hi CARLA Team,

we would like to report a problem we observed in the `Map.get_waypoint()` function in the Python API in CARLA version 0.9.9, when using it on a spiral geometry within a standalone OpenDRIVE .xodr map loaded via `config.py -x`.

We are using `Map.get_waypoint()` with `project_to_road = True`, `lane_type_flag` is default. Sporadically, when calling `Map.get_waypoint()` with the position of the ego vehicle as argument, the returned waypoint is not on the lane the vehicle is in, as expected, but instead on the adjacent left lane.
The problem does consistently occur at the same location in the map, which is inside a spiral geometry defined in the OpenDRIVE map. 

When inspecting the `lane_id` of the preceding correct waypoint and the first wrong waypoint, the `lane_id` changes to the value of the adjacent lane, although the vehicle clearly is close to the center of its original lane.

We appreciate your help!
Alex

germanros1987, 2020-06-26:
@Axel1092 please take a look at this.

Axel1092, 2020-07-02:
Hi @alexk-92580,
Can you please share more details about this issue? Does this happen within a junction? Does the spiral have a high curvature?

alexk-92580, 2020-07-03:
Hi @Axel1092,
thanks for taking care of this! The problem does not occur within a junction. Also, the curvatures are relatively low. We did observe, that the problem does not occur on all lanes. The spiral has 3 driveable lanes, and we have the issue on the rightmost driveable lane (ID -4) within a spiral with the following openDRIVE definition around 1629 meters into the segment:

 <geometry s="2500.0" x="2455.725103068875" y="-1198.2529509679275" hdg="-1.0" length="4981.650000000001">
                <spiral curvStart="0.0005" curvEnd="-0.0009860953494204312"/>
  </geometry>

If you need further information, please let me know!

Axel1092, 2020-07-03:
The problem is probably due to the discretization of the road in segments that enable the spatial queries and might be not fine enough. It would be great if you can provide a snippet of the geometry record of this spiral (or just its length, curvStart and curvEnd) to do some testing.

alexk-92580, 2020-07-13:
Hi @Axel1092 ,
the geometry record of the particular spiral in question is:
```
<geometry s="2500.0" x="2455.725103068875" y="-1198.2529509679275" hdg="-1.0" length="4981.650000000001">
                <spiral curvStart="0.0005" curvEnd="-0.0009860953494204312"/>
  </geometry>
```
Let me know if you need any more information!

Axel1092, 2020-07-15:
Hi @alexk-92580,
I found the problem, it was related to how we discretize the road in straight segment that was producing precision errors. A fix is placed in the related PR and will be included in the next release. Thank you for noticing this problem.

PR #3063 (2020-07-17): Fixed road discretization for long geometries.
#### Description

Problems with road discretization into straight segments were resulting in erroneous waypoints being retrieved by the function `get_waypoint()` .
This change sets an upper limit to road segments in order to improve precision for large geometries.

Fixes #2855

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 2, 3
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbacks
For very very large maps this may increase the amount of memory required to load the map.

---
This change is  [... cut]
closed 2020-07-17 by Axel1092