carla-simulator/carla #2858: Carla 0.9.9 crashes with Segmentation Fault error

ocean1211, 2020-05-15:
I use the Nvidia 1080Ti.
And the Nvidia driver for me 440, OS is 18.04LTS, OpenGL is 4.6
It always raises the error: signal 11 when I spawn the pedestrians.

Here is the detailed report:
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 772]
VulkanRHI::vkCreateBufferView(GetParent()->GetInstanceHandle(), &ViewInfo, VULKAN_CPU_ALLOCATOR, &View) failed, VkResult=-2
at /home/jenkins/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:1423
with error VK_ERROR_OUT_OF_DEVICE_MEMORY
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=148816 LargeMemoryPoolOffset=279952
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

Is there any solution?

ocean1211, 2020-05-15:
And actually I have faced the same problem with CARLA 0.9.7 and CARLA0.9.8 with the driver 390 and OpenGL3.1

ocean1211, 2020-05-15:
I use the compiled .tar.gz version of CARLA 0.9.9

tianli, 2020-05-17:
I had a similar crash when launching (Carla 0.99 binary installed via apt-get, Ubuntu 18.04). It looks like the GPU does not have enough memory to load all the assets. I have two GPUs on my Ubuntu 18.04, one with 2GB and another with 4GB memory. If I use the 2GB card for the primary display, Carla will crash (and without any meaningful error message). When I switch the 4GB card as the primary display, Carla can start without crashing. Checking with nvidia-smi, Carla uses 2.7 GB of video memory after launch.

ocean1211, 2020-05-18:
> I had a similar crash when launching (Carla 0.99 binary installed via apt-get, Ubuntu 18.04). It looks like the GPU does not have enough memory to load all the assets. I have two GPUs on my Ubuntu [...]

Thanks a lot. But it doesn't work for me...
The memory of my GPU is 11GB and the launching process does not fail.
When I try to spawn pedestrians, it often gets crashed. 
Are there any suggestions?

nuomizai, 2020-06-17:
> crash when launching (Carla 0.99 binary installed via apt-get, Ubuntu 18.04). It looks like the GPU does not have enough memory to load all the assets. I have two GPUs on my Ubuntu 18.04, one with [...]

Hi, @tianli . I met the same problem as that issue. How do you specify the GPU id?

tianli, 2020-06-17:
I didn't select, I just remove the 2GB card and move the 4GB ones to the spot. If you don't want to physically swap the card, maybe you can try this https://www.linuxbabe.com/desktop-linux/switch-intel-nvidia-graphics-card-ubuntu

germanros1987, 2020-06-26:
@bernatx could you help @ocean1211 with this issue?

bernatx, 2020-07-06:
@ocean1211 can you detail a bit more about how many pedestrians you are trying to spawn, and how many vehicles are in the scene?
Also, if you are using a custom map, did yu rebuild the binary file for pedestrians navigation?

ocean1211, 2020-07-06:
> @ocean1211 can you detail a bit more about how many pedestrians you are trying to spawn, and how many vehicles are in the scene?
> Also, if you are using a custom map, did yu rebuild the binary file for pedestrians navigation?

@bernatx I have tried to spawn 10 to 200 pedestrians randomly in every iteration. With the number of iterations increasing, the program will be stopped by this error. The number of vehicles is one.
I have not rebuild the binary file. And I use the compiled version for Linux and Town05.
Thanks a lot!

bernatx, 2020-07-06:
What do you mean by iterations?
You create and destroy pedestrians in the same script?
Are you sure you are destroying the pedestrians? Because there is an static limiation around 400 pedestrians and vehicles, altogether.
Probably we will resize this limitation in next version.

ocean1211, 2020-07-07:
> What do you mean by iterations?
> You create and destroy pedestrians in the same script?
> Are you sure you are destroying the pedestrians? Because there is an static limiation around 400 pedestrians and vehicles, altogether.
> Probably we will resize this limitation in next version.

Yes, I create and destroy pedestrians in the same script.
The program consists of many iterations.
In every iteration, pedestrians and the vehicle are spawned, and destroyed at the end of the iteration.
Sometimes when I don't spawn pedestrians in the environment, only spawning the vehicle repeatedly also raises the same error.

ozgurovic, 2020-07-26:
> Yes, I create and destroy pedestrians in the same script.
> The program consists of many iterations.
> In every iteration, pedestrians and the vehicle are spawned, and destroyed at the end of the [...]

I think I have a similar problem. The same script works for Town01, but for Town02, I get Segmentation fault (core dumped) error. I use around 18 vehicles and 15 pedestrians for spawning.

closed 2020-10-04 by stale[bot]

XieJinLiang, 2020-11-06:
And actually I have faced the same problem with CARLA 0.9.10 and CARLA0.9.9 with the nvidisa driver 450.66

yasser-h-khalil, 2020-11-14:
Any solution regarding this? I am facing the same error! version 0.9.10

XieJinLiang, 2020-11-14:
@yasser-h-khalil 
I have work in docker running 0.9.10， refer to #3555

muety, 2023-12-20:
I'm getting the same error now with 0.9.15 and NVIDIA driver 545.23. Got plenty of GPU RAM.