carla-simulator/carla #2862: Make error with Ubuntu 1.04 with Carla 0.9.9

BolinZhou, 2020-05-17:
building Carla 0.9.9 on Ubuntu 18.04
it got stuck on `make launch`

```
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c2-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: CARLA version 0.9.9-56-g09ec923c.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
-- The C compiler identification is Clang 8.0.1
-- The CXX compiler identification is Clang 8.0.1
-- Could not determine Eclipse version, assuming at least 3.6 (Helios). Adjust CMAKE_ECLIPSE_VERSION if this is wrong.
-- Check for working C compiler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build debug:   OFF
-- Build release: ON
-- Build test:    ON
-- Configuring done
CMake Warning in CMakeLists.txt:
  The build directory is a subdirectory of the source directory.

  This is not supported well by Eclipse.  It is strongly recommended to use a
  build directory which is a sibling of the source directory.

-- Generating done
-- Build files have been written to: /home/bolin/carla/Build/libcarla-server-build.release
[2/52] Building CXX object LibCarla/cmake/server/.../carla_server.dir/__/__/source/carla/Buffer.cpp.o
FAILED: LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Buffer.cpp.o 
/usr/bin/clang++-8  -DASIO_NO_EXCEPTIONS -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_EXCEPTIONS -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_NO_EXCEPTIONS -DLIBCARLA_TEST_CONTENT_FOLDER=\"/home/bolin/carla/Build/test-content\" -DPUGIXML_NO_EXCEPTIONS -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /home/bolin/carla/Build/llvm-8.0-install/include/c++/v1 -fno-exceptions -fno-rtti -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /home/bolin/carla/Build/llvm-8.0-install/include/c++/v1 -fno-exceptions -fno-rtti   -O3 -DNDEBUG -MD -MT LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Buffer.cpp.o -MF LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Buffer.cpp.o.d -o LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Buffer.cpp.o -c ../../LibCarla/source/carla/Buffer.cpp
In file included from ../../LibCarla/source/carla/Buffer.cpp:1:
../../LibCarla/cmake/../source/carla/Buffer.h:320:42: error: no member named 'is_const_buffer_sequence' in namespace 'boost::asio'
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                            ~~~~~~~~~~~~~^
../../LibCarla/cmake/../source/carla/Buffer.h:320:67: error: 'T' does not refer to a value
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                  ^
../../LibCarla/cmake/../source/carla/Buffer.h:319:24: note: declared here
    template <typename T>
                       ^
../../LibCarla/cmake/../source/carla/Buffer.h:320:71: error: member 'value' declared as a template
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                      ^
../../LibCarla/cmake/../source/carla/Buffer.h:320:76: error: expected ';' at end of declaration list
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                           ^
                                                                           ;
../../LibCarla/cmake/../source/carla/Buffer.h:111:9: error: use of undeclared identifier 'ReuseThisBuffer'
        ReuseThisBuffer();
        ^
../../LibCarla/cmake/../source/carla/Buffer.h:309:7: error: no matching member function for call to 'copy_from'
      copy_from(0u, data, size);
      ^~~~~~~~~
../../LibCarla/cmake/../source/carla/Buffer.h:302:10: note: candidate function template not viable: requires single argument 'source', but 3 arguments were provided
    void copy_from(const T &source) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:314:10: note: candidate function not viable: requires 2 arguments, but 3 were provided
[... 679 lines cut ...]
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:314:10: note: candidate function not viable: requires 2 arguments, but 3 were provided
    void copy_from(size_type offset, const Buffer &rhs) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:308:10: note: candidate function not viable: requires 2 arguments, but 3 were provided
    void copy_from(const value_type *data, size_type size) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:315:7: error: no matching member function for call to 'copy_from'
      copy_from(offset, rhs.buffer());
      ^~~~~~~~~
../../LibCarla/cmake/../source/carla/Buffer.h:314:10: note: candidate function not viable: no known conversion from 'boost::asio::const_buffer' to 'const carla::Buffer' for 2nd argument
    void copy_from(size_type offset, const Buffer &rhs) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:308:10: note: candidate function not viable: no known conversion from 'carla::Buffer::size_type' (aka 'unsigned int') to 'const carla::Buffer::value_type *' (aka 'const unsigned char *') for 1st argument
    void copy_from(const value_type *data, size_type size) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:302:10: note: candidate function template not viable: requires single argument 'source', but 2 arguments were provided
    void copy_from(const T &source) {
         ^
7 errors generated.
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:120: recipe for target 'LibCarla.server.release' failed
make: *** [LibCarla.server.release] Error 1

```

BolinZhou, 2020-05-18:
It passed with miracle.  I performed `make clean` then fixed. 

Thanks

closed 2020-05-18 by BolinZhou

gumpcarl, 2021-04-29:
wrong boost version