carla-simulator/carla #2863: Carla obstacle detection sensor problem

cokeyke, 2020-05-18:
This is my code and my distance output from the sensor is always 0.
class ObstacleDetectionSensor(object):
    def __init__(self, parent_actor):
        self.sensor = None
        self._parent = parent_actor
        self.distance = None
        self._event_count = 0
        world = self._parent.get_world()
        bp = world.get_blueprint_library().find('sensor.other.obstacle')
        bp.set_attribute('distance','250')
        bp.set_attribute('hit_radius','250')
        bp.set_attribute('debug_linetrace','true')
        bp.set_attribute('only_dynamics', 'true')
        self.sensor = world.spawn_actor(bp, carla.Transform(), attach_to=self._parent)
        # We need to pass the lambda a weak reference to self to avoid circular
        # reference.
        weak_self = weakref.ref(self)
        self.sensor.listen(lambda event: ObstacleDetectionSensor._on_obstacle(weak_self, event))

    @staticmethod
    def _on_obstacle(weak_self, event):
        self = weak_self()
        if not self:
            return
        self.distance = event.distance
        self._event_count += 1
        print ("Event %s, in line of sight with %s at distance %u" % (self._event_count, event.other_actor.type_id, event.distance))

germanros1987, 2020-06-26:
@glopezdiest could you provide some help with this?

glopezdiest, 2020-06-26:
Hey @cokeyke. I understand that even though the distance is 0, the other_actor is indeed an actor, correct? Could you tell me the CARLA version you are using so that I can test myself?

Shubham-Sahoo, 2020-07-30:
@glopezdiest  I'm having the same issue and I'm using CARLA version 0.9.7.

fabianoboril, 2020-07-31:
@glopezdiest: I can confirm this issue with current master.

fabianoboril, 2020-07-31:
I did some further testing to understand how this sensor works. It seems to perform sphere tracing, which is pretty well explained here: https://www.youtube.com/watch?v=Sg1siSsT2-0
Hence, the HitRadius is the important parameter to think about. A HitRadius of 250m will create a gigantic sphere, which will almost inevitably cause collisions. I am now using a HitRadius of just 1m, and now things are working much better.
Hope this helps.

glopezdiest, 2020-08-03:
Yes, exactly what @fabianoboril said. While the distance is important, the *hit_radius* must be small in order to avoid unwanted collisions. For example, by putting it to 5m, the *hit_radius* is large enough to detect the road as a possible collision, so the event will always return the road, with distance equal to zero.

This doesn't seem to be correctly reflected at the docs, so I'm adding @sergi-e, in case he thinks some modifications are needed.

closed 2020-08-03 by glopezdiest

matteB10, 2022-08-12:
I try to use the obstacle detector, but it does not ever seem to register cars or walkers as obstacles (even when they are right in front). The obstacles detected are 'Road', 'Terrain' (with distance 0.0m) and very occasionally a traffic light pole or something. I've tried many different values for the 'hit_radius', as small as 0.25m, but I am still getting obstacle Road/Terrain/RoadLine with 0.0 distance.

I have also tried to enable 'debug_linetrace' to get more information of the tracing area, but it does not work either.  
I am using CARLA 0.9.13. Any suggestions?

aagrawa3-dev, 2022-10-23:
> I try to use the obstacle detector, but it does not ever seem to register cars or walkers as obstacles (even when they are right in front). The obstacles detected are 'Road', 'Terrain' (with distance 0.0m) and very occasionally a traffic light pole or something. I've tried many different values for the 'hit_radius', as small as 0.25m, but I am still getting obstacle Road/Terrain/RoadLine with 0.0 distance.
> 
> If 'only_dynamics' is set to true, it can detect other cars. But how can I make the obstacle detector work for both dynamic and static objects?
> 
> I have also tried to enable 'debug_linetrace' to get more information of the tracing area, but it does not show. I am using CARLA 0.9.13. Any suggestions?

Also, faced same question: How can we make dynamic and non-dynamic obstacles detected(without detecting road as obstacle) in Carla.

ivevasiljevic, 2023-05-11:
Facing the same issue. No matter what I change, I always get 0.0 m.

howicc, 2024-01-05:
Did you solve this?
Using 0.9.14 and the distance is always 0.0 no matter where the detected object is. When debugging the ObstacleDetectionEvent object, the detected actor seems to be correct but location of the actor is attached to does not have data. Could this be the reason?

other_actor.bounding_box:

BoundingBox(Location(x=0.029219, y=0.000000, z=0.735861), Extent(x=2.395890, y=1.081725, z=0.743830), Rotation(pitch=0.000000, yaw=0.000000, roll=0.000000))

actor.bounding_box:
BoundingBox(Location(x=-nan, y=-nan, z=-nan), Extent(x=-inf, y=-inf, z=-inf), Rotation(pitch=0.000000, yaw=0.000000, roll=0.000000))