carla-simulator/carla #2864: Recorder file info - can't get all the data.

najibghadri, 2020-05-18:
I can't get any data of the vehicles and pedestrians from a recorder log. I have read through the API, used all arguments etc, and even the code leading to CarlaRecorderQuery.cpp.
I have a recording full of cars aroung the hero car and pedestrians and later I call client.show_recorder_file_info(recorder_filename, show_all) with the server on, and I get a header, listing all the Creates of the sensors and traffic lights and whatnot, but only one vehicle the hero vehicle, then it lists the frames list of frames but still no data about any pedestrian or vehicle.

Why do I need this is simple: I need ground truth data for testing and training NNs. That's mainly why I am using a simulator.

A snapshot so you see there are vehicles:
[image]
The result:
 
https://gist.github.com/najibghadri/285edb21dc44343a141f8cccea45f440#file-show_recorder_file_info-py-a-output 

As I said I tried everything, tried show_recorder_actors_blocked min distance min time with very small, zero, -1. Hope I am missing something. Obviously tried opening the file too.

germanros1987, 2020-06-26:
This sounds weird. It may be a bug. @bernatx @Axel1092 could you take a look?

bernatx, 2020-07-02:
This is very strange. Did you use the recorder from the **manual_control.py** script, or you used separated scripts as **start_recording.py**?
We will try to reproduce that, but it is clear that the recorder only got the hero vehicle.
Can you specify the steps you follow to record that session?
Thanks.

closed 2020-09-07 by stale[bot]

rajesh-gurgoan, 2023-06-26:
@germanros1987 @bernatx I also have been facing the same issue. I had update the start_recording.py with calling the client method 'show_recorder_file_info'  at the end.  I called the client method like this

client.show_recorder_file_info(<recorded_file_name>, True).

Below is the text, I got

Version: 1
Map: Town10HD_Opt
Date: 06/26/23 12:00:37

Frame 1 at 0 seconds
 Create 1: spectator (0) at (-17219.7, 18385.9, 2763.81)
 Create 2: traffic.stop (4) at (5745, 6115, 15)
 Create 3: traffic.stop (4) at (2610, 7515, 15)
 Create 4: traffic.stop (4) at (3535, 5360, 15)
 Create 5: traffic.yield (4) at (4960, 3850, 15)
 Create 6: traffic.unknown (4) at (4315, 6645, 0)
 Create 7: traffic.unknown (4) at (4404.62, 3779.89, 0)
 Create 8: traffic.unknown (4) at (-2815, 6602.97, 0)
 Create 9: traffic.traffic_light (3) at (-11923, 509.315, 26.277)
 Create 10: traffic.traffic_light (3) at (11545, 3504.49, 22.2726)
 Create 11: traffic.traffic_light (3) at (-4617.16, -7355.66, 25.4254)
 Create 12: traffic.traffic_light (3) at (-3444.92, -5102.05, 25.4255)
 Create 13: traffic.traffic_light (3) at (11445.2, 2119.6, 25.42)
 Create 14: traffic.traffic_light (3) at (8979.19, 2092.21, 25.42)
 Create 15: traffic.traffic_light (3) at (-3193.08, 2030.2, 25.4254)
 Create 16: traffic.traffic_light (3) at (-6426.42, 706.331, 25.4254)
 Create 17: traffic.traffic_light (3) at (-6235.21, 2019.69, 25.4254)
 Create 18: traffic.traffic_light (3) at (-9492.82, 2033.48, 25.4256)
 Create 19: traffic.traffic_light (3) at (-5924.23, -5149.84, 25.4254)
 Create 20: traffic.traffic_light (3) at (-11923.8, 1900.45, 25.4253)
 Create 21: traffic.traffic_light (3) at (-5871.54, 14582.5, 25.4188)
 Create 22: traffic.traffic_light (3) at (-6209.93, 12375.6, 28.3086)
 Create 23: traffic.traffic_light (3) at (-3163.64, 3358.93, 25.4254)
Frame 12 at 0.0855917 seconds
 Create 24: vehicle.dodge.charger_2020 (1) at (5740.36, 2834.35, 60)
  role_name = autopilot
  color = 0,0,0
  sticky_control = true
  terramechanics = false
  object_type = 
  generation = 2
  base_type = car
  special_type = 
  number_of_wheels = 4
  has_dynamic_doors = true
  has_lights = true
 Create 25: vehicle.jeep.wrangler_rubicon (1) at (9907.86, 4214.18, 60)
  role_name = autopilot
  color = 168,0,27
  sticky_control = true
  terramechanics = false
  object_type = 
  generation = 1
  base_type = car
  special_type = 
  number_of_wheels = 4
  has_dynamic_doors = false
  has_lights = false
 Create 26: vehicle.bmw.grandtourer (1) at (5312.27, 13394.7, 60.0029)
  role_name = autopilot
  color = 12,38,88
  sticky_control = true
  terramechanics = false
  object_type = 
  generation = 1
  base_type = 
  special_type = 
  number_of_wheels = 4
  has_dynamic_doors = false
  has_lights = false
 Create 27: vehicle.volkswagen.t2 (1) at (5756.83, -6784.99, 60)
  role_name = autopilot
  color = 0,33,5
  sticky_control = true
  terramechanics = false
  object_type = 
  generation = 1
  base_type = van
  special_type = 
  number_of_wheels = 4
  has_dynamic_doors = false
  has_lights = true
 Create 28: vehicle.kawasaki.ninja (1) at (-5218.67, 4256.51, 60)
  role_name = autopilot
  color = 11,129,0
  driver_id = 0
  base_type = 
  sticky_control = true
  terramechanics = false
  object_type = 
  generation = 1
  special_type = motorcycle
  number_of_wheels = 2
  has_dynamic_doors = false
  has_lights = false
 Create 29: vehicle.tesla.cybertruck (1) at (-7803.42, 1296.72, 60)
  role_name = autopilot
  sticky_control = true
  terramechanics = false
  object_type = 
  generation = 2
  base_type = truck
  special_type = electric
  number_of_wheels = 4
  has_dynamic_doors = false
  has_lights = true
 Create 30: vehicle.audi.tt (1) at (-1710.54, 1325.75, 60)
  role_name = autopilot
  color = 33,33,33
  sticky_control = true
  terramechanics = false
  object_type = 
  generation = 1
  base_type = car
  special_type = 
  number_of_wheels = 4
  has_dynamic_doors = false
  has_lights = true
 Create 31: vehicle.kawasaki.ninja (1) at (10994.7, -1718.8, 59.9999)
  role_name = autopilot
  color = 0,0,0
  driver_id = 0
  base_type = 
  sticky_control = true
  terramechanics = false
  object_type = 
  generation = 1
  special_type = motorcycle
  number_of_wheels = 2
  has_dynamic_doors = false
  has_lights = false
 Create 32: vehicle.citroen.c3 (1) at (10950.3, 5329.32, 60)
  role_name = autopilot
  color = 16,44,21
  sticky_control = true
  terramechanics = false
  object_type = 
  generation = 1
  base_type = car
  special_type = 
  number_of_wheels = 4
  has_dynamic_doors = false
  has_lights = false
 Create 33: vehicle.tesla.model3 (1) at (10256.6, 4396.57, 60)
  role_name = autopilot
  color = 180,42,42
  sticky_control = true
  terramechanics = false
  object_type = 
  generation = 1
  base_type = car
  special_type = electric
  number_of_wheels = 4
  has_dynamic_doors = false
  has_lights = true
Frame 19820 at 160.36 seconds
 Destroy 24
 Destroy 25
 Destroy 26
 Destroy 27
 Destroy 28
 Destroy 29
 Destroy 30
 Destroy 31
 Destroy 32
 Destroy 33

Frames: 19820
Duration: 160.36 seconds

minhaj6, 2024-10-08:
> This is very strange. Did you use the recorder from the **manual_control.py** script, or you used separated scripts as **start_recording.py**? We will try to reproduce that, but it is clear that the [...]

Hi, I am facing the same issue. I used `start_recording.py` to generate the test1.log file.