carla-simulator/carla #2865: Build Carla on Windows error

AdelinaK, 2020-05-18:
I tried to build Carla on Windows and after executing the make launch command I've got this error. I'm new to this and maybe I did something wrong. Any ideas?

-[BuildCarlaUE4]: Building Unreal Editor...
Creating makefile for CarlaUE4Editor (no existing makefile)
Parsing headers for CarlaUE4Editor
  Running UnrealHeaderTool "C:\carla\Unreal\CarlaUE4\CarlaUE4.uproject" "C:\carla\Unreal\CarlaUE4\Intermediate\Build\Win64\CarlaUE4Editor\Development\CarlaUE4Editor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
Error: Couldn't find parent type for 'CarlaLightSubsystem' named 'UWorldSubsystem' in current module or any other module parsed so far.
make: *** [Util/BuildTools/Windows.mk:20: CarlaUE4Editor] Error 5

closed 2020-05-21 by AdelinaK

RehabHsn, 2020-05-24:
have you found a solution??

RehabHsn, 2020-05-24:
I have the same error

oeichenwei, 2020-08-13:
i resolved the problem by upgrade UE to 4.24