carla-simulator/carla #2866: semantic segmentation does not work properly on props ingestion- version 0.9.8 linux from source

omrizen, 2020-05-19:
Hello
working on carla release 0.9.8 on linux from source
after prop ingestion (using json file with tag e.g. -  {  
  "maps": [
  ],"props": [
    {      "name": "MyProp01",      "size": "big",      "source": "./props/prop01/prop01.fbx",      "tag": "Vehicles"} ]} )
the image that I get with semantic segmentation sensor colores the object in black (so my tag has no effect). 
I found out the "render custom" is disabled on  my prop after start playing , comparing to other constant actors in the scene,  so perhaps this should be fixed

Thanks

germanros1987, 2020-06-26:
Did you try with 0.9.9? Any differences?

Jabkowsky, 2020-07-16:
I have similar issue on 0.9.9 version
See: https://github.com/carla-simulator/carla/issues/3070

PR #3088 (2020-07-23): Fixed untagged objects spawned from the client
#### Description
Quick fix for an error that prevent objects from being correctly tagged after being spawned from the client side.

Fixes #2866

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 3
  * **Unreal Engine version(s):** 2.24

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3088)
closed 2020-07-23 by bernatx