carla-simulator/carla #2868: Spawn camera to look a specific actor

JWHennessey, 2020-05-19:
Hi,

I'd like to spawn a camera that will look at a specific actor from a given position. 

Theoretically what I am trying to do is generate a location which is a given offset to the camera. Then using a lookAt function calculate camera matrix to look at the target actor. Then from the 3x3 matrix then extract the pitch, yaw, roll to set the camera transformation. 

Here is the code below, any suggestion very welcome. 

```
def lookAt(eye, at, up):
    zaxis = (eye - at)
    zaxis = zaxis / np.linalg.norm(zaxis)
    xaxis = np.cross(zaxis, up)
    xaxis = xaxis / np.linalg.norm(xaxis)

    yaxis = np.cross(xaxis, zaxis)
    yaxis = yaxis / np.linalg.norm(yaxis)

    m = np.zeros((4, 4))
    m[0, :-1] = xaxis
    m[1, :-1] = yaxis
    m[2, :-1] = zaxis
    m[:-1, 3] = eye
    m[3, 3] = 1.0
    return m

car_loc = self.car.get_location()
cam_position = np.array([car_loc.x + 10, car_loc.y + 0.1, car_loc.z + 2])

target = np.array([car_loc.x, car_loc.y, car_loc.z])

up = np.array([0, 0, 1])
viewmatrix = lookAt(cam_position, target, up) 

euler = rot.as_euler('XYZ', degrees=True)  

camera_transform = carla.Transform(
            carla.Location(cam_position[0], cam_position[1], cam_position[2]),
            carla.Rotation(yaw=euler[2], pitch=euler[1], roll=euler[0]))

self.camera.set_transform(camera_transform)

```

germanros1987, 2020-06-26:
@JWHennessey the general idea of your code makes sense. I guess you just need some help to get the coordinate system right. @marcgpuig could you help with that?

marcgpuig, 2020-06-30:
Hi @JWHennessey,
I think this may be related to https://github.com/carla-simulator/carla/issues/2915#issuecomment-651726825, let's wait until #2997 is merged so you can test it with the provided matrix.

closed 2020-09-05 by stale[bot]