carla-simulator/carla #2870: sensor.other.imu not found

a4zhangfei, 2020-05-20:
There is no sensor in my blueprint other.imu and sensor.other.radar
`len(bp.filters('sensor.other.*')` is 4

germanros1987, 2020-06-26:
Could you provide some information...? Version of CARLA, OS, etc..?

a4zhangfei, 2020-06-27:
> Could you provide some information...? Version of CARLA, OS, etc..?

OS : `ubuntu 18.04`
Version of CARLA: `0.9.6 release`
The version I use is not compiled. But many methods and properties are missing. For example, `waypoint`'s method `previous_ until_ lane_ start`，`sensor.other.imu`

germanros1987, 2020-06-27:
IMU and RADAR were not available in 0.9.6. They were introduced in more recent versions of CARLA. I would recommend you to upgrade to 0.9.9.

closed 2020-06-27 by germanros1987