carla-simulator/carla #2871: ImportError: libproj.so.9

vishal-246, 2020-05-20:
Hi for 0.9.9.1-RSS-ready compiled patch release i'm facing the `libproj.so.9` import error with all the PythonAPI's 
````
Traceback (most recent call last):
  File "./config.py", line 28, in <module>
    import carla
  File "../carla/dist/carla-0.9.9-py2.7-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
  File "../carla/dist/carla-0.9.9-py2.7-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
  File "../carla/dist/carla-0.9.9-py2.7-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
ImportError: libproj.so.9: cannot open shared object file: No such file or directory
```
`
please help me to resolve this. 
and with 0.9.9 version API there was no such errors. and python version is used same for both the variant i.e 2.7.

Thanks.

germanros1987, 2020-06-26:
@berndgassmann could you please have a look?

berndgassmann, 2020-06-26:
Hmm, I would guess the issue is not related to the delta within 0.9.9 vs 0.9.9.1, but rather due to the fact that you now want to use the RSS variant.
The egg was built using ubuntu 16.04.
If you have Ubuntu 16.04 you can solve the issue easily by installation of https://packages.ubuntu.com/xenial/libproj9.
If you use Ubuntu 18.04 the problem is that there another version of the library is provided: https://packages.ubuntu.com/bionic/libproj12
So you'll have to build CARLA with RSS support on your own for now, I fear.

So I'll try to solve the issue by linking the library statically to come to a real solution to this issue, but this will take some time.

PR #3024 (2020-07-14): Update to AD RSS Library 4.0.0
#### Description

- Added support for unstructured scenes and pedestrians.
- Rework rss example folder structure
- Python based debug drawing
- Handle negative speeds
- Fixes #2871 by linking libproj statically.
- Fixes #2958 by using sequential colcon build
- Allow setting log level
- Add ActorConstellationCallback
- Move setting of RssDynamics to python code

#### Where has this been tested?

  * **Platform(s):** ...
  * **Python version(s):** ...
  * **Unreal Engine version(s):** ...

#### Possible Drawbacks

---
This change is [[image]](https://reviewable.io/ [... cut]
commit 88ae9d04 (2020-07-14): Update to AD RSS Library 4.0.0

- Added support for unstructured scenes and pedestrians.
- Rework rss example folder structure
- Python based debug drawing
- Handle negative speeds
- Fixes #2871 by linking libproj statically.
- Fixes #2958 by using sequential colcon build
- Allow setting log level
- Add ActorConstellationCallback
- Move setting of RssDynamics to python code
closed 2020-07-14 by marcgpuig