carla-simulator/carla #2876: How to get the action autopilot would have taken?

GershomA, 2020-05-21:
Hi, I'm using carla 0.99 on windows.
I'm trying to train a neural network to drive a car, but use the autopilot as expert labels for what the neural network should have done.
Is it possible in the new api to query the autopilot system for an action, but not actually take that action on the vehicle?
This used to be possible in the 0.8.x API, but I'm not seeing a way to do so in 0.9.9

jackbart94, 2020-05-22:
Hi,

The autopilot has been deprecated from 0.9.7, now we are using the Traffic Manager. You can't query it yet for actions that are going to be taken, but we'll make that available soon. However, you can try doing it yourself.

GershomA, 2020-05-22:
Ah I see, thanks. I'll try to find a workaround

closed 2020-05-22 by GershomA