carla-simulator/carla #2878: Can not change map in Carla 0.9.9 windows

Shiyunyang-seu, 2020-05-22:
Hi, I want to use the Carla 0.9.9 to make co-simulation with sumo.
But, I tried the config.py -m Town04 and also followed the tutorial https://carla.readthedocs.io/en/latest/core_map/#changing-the-map, but a fatal error both occurs.(said carlaUE4 crashed with something...) The 0.9.5 version is also tested and the map(Town01,02...) can be changed without error.
I tested in two different PCs in windows10.

germanros1987, 2020-06-26:
This is strange. @bernatx could you have a look?

bernatx, 2020-07-02:
Hi,
This problem was already fixed in 0.9.9.4. It was caused by a workaround about one bug in UE4, that officially released a patch also. In CARLA 0.9.9.4 we removed the workaround and release a patch for the UE 4.24. The patch for UE 4.24 is only needed for Ubuntu machines, related to the Vulkan driver.

That would solve your problem.

closed 2020-09-07 by stale[bot]