carla-simulator/carla #2879: Problem with starting the simulation in UnrealEngine after completing Windows build

NicoSchoenfisch, 2020-05-22:
Hey everyone,

so I have a problem right now that might be pretty basic, but I didn't find a solution in the docs for it.

I followed the instructions to build Carla on Windows step by step from here: https://carla.readthedocs.io/en/latest/build_windows/
and am now stuck at the very last step.

'make launch' and 'make PythonAPI' both finished successfully, and the project is also opened in UnrealEngine and looks like this:

[image]

citing from the instructions, my steps after 'make launch' should be this:

"make launch compiles the server simulator and launches Unreal Engine. Press Play to start the spectator view and close the editor window to exit. Camera can be moved with WASD keys and rotated by clicking the scene while moving the mouse around."

Now under the "Play" Tab in Unreal Engine, I can choose a mode. If I choose anything other than "Standalone Game", I get this message:

[image]

If I choose to play anyway, this is what it looks like:

[image]

On the other hand, if I choose "Standalone Game" as the mode, I don't get the warning or an error message, but I still end up with the black screen and this grey stick-thingy in the middle, like this:

[image]

From what I understand from reading the instructions and watching some videos, by pressing "Play" at this point after the build, I should be able to start a simulation of the current map (Town03 as standard) in which I can move the camera with WASD, instead of having this black screen. It's definitely possible I missed something I have to do before it works, but I have no idea what it is.

Could someone maybe help me out with this?

Thanks in advance!

Shiyunyang-seu, 2020-05-22:
Hi, I also followed the tutorial https://carla.readthedocs.io/en/latest/build_windows/. I want to ask how to use these two commands: 'make launch' and 'make PythonAPI' , I also download the Gnuwin32, but in my cmd, it said 'make' is not an internal or external command, nor is it a runnable program.

germanros1987, 2020-06-26:
@bernatx you are probably the right person for this task.

bernatx, 2020-07-06:
Hi @NicoSchoenfisch did you download all the content assets? It seems that you don't have it.
Refer to this link with instructions: https://carla.readthedocs.io/en/latest/build_update/

closed 2020-09-12 by stale[bot]