carla-simulator/carla #2880: Build cala 0.9.9,  Make PythonAPI error

Avens666, 2020-05-23:
Hi
I build carla 0.9.9 on Win10 + VS2017 + UE4.24 +Python38
I run **make launch** to build server succeed. and I can use UE4 package carla.

But **make Python** failed
There is error when link : 
libboost_filesystem-vc141-mt-x64-1_72.lib(exception.obj) : error LNK2001: Unresolved external symbols __CxxFrameHandler4
libboost_filesystem-vc141-mt-x64-1_72.lib(codecvt_error_category.obj) : error LNK2001: Unresolved external symbols __CxxFrameHandler4
libboost_filesystem-vc141-mt-x64-1_72.lib(directory.obj) : error LNK2001: Unresolved external symbols __CxxFrameHandler4
libboost_filesystem-vc141-mt-x64-1_72.lib(path_traits.obj) : error LNK2001: Unresolved external symbols __CxxFrameHandler4
libboost_filesystem-vc141-mt-x64-1_72.lib(operations.obj) : error LNK2001: Unresolved external symbols __CxxFrameHandler4
libboost_filesystem-vc141-mt-x64-1_72.lib(path.obj) : error LNK2001: Unresolved external symbols __CxxFrameHandler4
libboost_filesystem-vc141-mt-x64-1_72.lib(exception.obj) : error LNK2001: Unresolved external symbols __GSHandlerCheck_EH4
libboost_filesystem-vc141-mt-x64-1_72.lib(codecvt_error_category.obj) : error LNK2001: Unresolved external symbols __GSHandlerCheck_EH4
libboost_filesystem-vc141-mt-x64-1_72.lib(directory.obj) : error LNK2001: Unresolved external symbols __GSHandlerCheck_EH4
libboost_filesystem-vc141-mt-x64-1_72.lib(path_traits.obj) : error LNK2001: Unresolved external symbols __GSHandlerCheck_EH4
libboost_filesystem-vc141-mt-x64-1_72.lib(operations.obj) : error LNK2001: Unresolved external symbols __GSHandlerCheck_EH4
libboost_filesystem-vc141-mt-x64-1_72.lib(path.obj) : error LNK2001: Unresolved external symbols __GSHandlerCheck_EH4

anyone had same error like me?   How to fix it.  Thank !

It is link command
'''
"d:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\link.exe" /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:dependencies/lib /LIBPATH:D:\Tools\Python383\libs /LIBPATH:D:\Tools\Python383\PCbuild\amd64 "/LIBPATH:d:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\lib\x64" "/LIBPATH:d:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\um\x64" "/LIBPATH:d:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\lib\x64" "/LIBPATH:d:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\um\x64" /EXPORT:PyInit_libcarla build\temp.win-amd64-3.8\Release\source/libcarla/libcarla.obj build\in38\Buffer.obj build\in38\Exception.obj build\in38\FileSystem.obj build\in38\StringUtil.obj build\in38\client\Actor.obj build\in38\client\ActorAttribute.obj build\in38\client\ActorBlueprint.obj build\in38\client\ActorList.obj build\in38\client\BlueprintLibrary.obj build\in38\client\DebugHelper.obj build\in38\client\Junction.obj build\in38\client\LaneInvasionSensor.obj build\in38\client\Light.obj build\in38\client\LightManager.obj build\in38\client\Map.obj build\in38\client\ServerSideSensor.obj build\in38\client\TimeoutException.obj build\in38\client\TrafficLight.obj build\in38\client\TrafficSign.obj build\in38\client\Vehicle.obj build\in38\client\Walker.obj build\in38\client\WalkerAIController.obj build\in38\client\Waypoint.obj build\in38\client\World.obj build\in38\client\detail\ActorFactory.obj build\in38\client\detail\ActorState.obj build\in38\client\detail\ActorVariant.obj build\in38\client\detail\Client.obj build\in38\client\detail\Episode.obj build\in38\client\detail\EpisodeProxy.obj build\in38\client\detail\EpisodeState.obj build\in38\client\detail\Simulator.obj build\in38\client\detail\WalkerNavigation.obj build\in38\geom\GeoLocation.obj build\in38\geom\Math.obj build\in38\geom\Mesh.obj build\in38\geom\MeshFactory.obj build\in38\nav\Navigation.obj build\in38\nav\WalkerEvent.obj build\in38\nav\WalkerManager.obj build\in38\opendrive\OpenDriveParser.obj build\in38\opendrive\parser\ControllerParser.obj build\in38\opendrive\parser\GeometryParser.obj build\in38\opendrive\parser\GeoReferenceParser.obj build\in38\opendrive\parser\JunctionParser.obj build\in38\opendrive\parser\LaneParser.obj build\in38\opendrive\parser\ObjectParser.obj build\in38\opendrive\parser\ProfilesParser.obj build\in38\opendrive\parser\RoadParser.obj build\in38\opendrive\parser\SignalParser.obj build\in38\opendrive\parser\TrafficGroupParser.obj build\in38\pointcloud\PointCloudIO.obj build\in38\road\Lane.obj build\in38\road\LaneSection.obj build\in38\road\Map.obj build\in38\road\MapBuilder.obj build\in38\road\MapData.obj build\in38\road\Road.obj build\in38\road\SignalType.obj build\in38\road\element\Geometry.obj build\in38\road\element\LaneCrossingCalculator.obj build\in38\road\element\LaneMarking.obj build\in38\road\element\Waypoint.obj build\in38\rpc\WeatherParameters.obj build\in38\sensor\Deserializer.obj build\in38\sensor\data\LaneInvasionEvent.obj build\in38\sensor\s11n\CollisionEventSerializer.obj build\in38\sensor\s11n\DVSEventArraySerializer.obj build\in38\sensor\s11n\EpisodeStateSerializer.obj build\in38\sensor\s11n\GnssSerializer.obj build\in38\sensor\s11n\ImageSerializer.obj build\in38\sensor\s11n\IMUSerializer.obj build\in38\sensor\s11n\LidarSerializer.obj build\in38\sensor\s11n\NoopSerializer.obj build\in38\sensor\s11n\ObstacleDetectionEventSerializer.obj build\in38\sensor\s11n\RadarSerializer.obj build\in38\sensor\s11n\SensorHeaderSerializer.obj build\in38\streaming\detail\Dispatcher.obj build\in38\streaming\detail\StreamStateBase.obj build\in38\streaming\detail\Token.obj build\in38\streaming\detail\tcp\Client.obj build\in38\streaming\detail\tcp\Server.obj build\in38\streaming\detail\tcp\ServerSession.obj build\in38\trafficmanager\BatchControlStage.obj build\in38\trafficmanager\CollisionStage.obj build\in38\trafficmanager\InMemoryMap.obj build\in38\trafficmanager\LocalizationStage.obj build\in38\trafficmanager\LocalizationUtils.obj build\in38\trafficmanager\MotionPlannerStage.obj build\in38\trafficmanager\Parameters.obj build\in38\trafficmanager\PerformanceDiagnostics.obj build\in38\trafficmanager\PIDController.obj build\in38\trafficmanager\PipelineStage.obj build\in38\trafficmanager\SimpleWaypoint.obj build\in38\trafficmanager\TrafficLightStage.obj build\in38\trafficmanager\TrafficManager.obj build\in38\trafficmanager\TrafficManagerLocal.obj build\in38\trafficmanager\TrafficManagerRemote.obj /OUT:build\lib.win-amd64-3.8\carla\libcarla.cp38-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.8\Release\source/libcarla\libcarla.cp38-win_amd64.lib shlwapi.lib G:\UE4\carla17\PythonAPI\carla\dependencies/lib\carla_client.lib G:\UE4\carla17\PythonAPI\carla\dependencies/lib\Detour.lib G:\UE4\carla17\PythonAPI\carla\dependencies/lib\DetourCrowd.lib G:\UE4\carla17\PythonAPI\carla\dependencies/lib\libboost_filesystem-vc141-mt-x64-1_72.lib G:\UE4\carla17\PythonAPI\carla\dependencies/lib\libboost_python38-vc141-mt-x64-1_72.lib G:\UE4\carla17\PythonAPI\carla\dependencies/lib\libpng.lib G:\UE4\carla17\PythonAPI\carla\dependencies/lib\Recast.lib G:\UE4\carla17\PythonAPI\carla\dependencies/lib\rpc.lib G:\UE4\carla17\PythonAPI\carla\dependencies/lib\zlib.lib
'''

Krxtopher, 2020-05-27:
I have encountered this issue myself and have not found a workaround (besides switching to a different computer). This problem was also reported by another user in this issue: https://github.com/carla-simulator/carla/issues/2741

Avens666, 2020-05-28:
Thank you for your reply.
I try 2 computer, one had VS2019 and VS2017, one only has VS2017. but had the same problem.
maybe  try another computer or wait next calar version.
now I use the python lib in release version 0.9.9.

germanros1987, 2020-06-26:
@bernatx you are our Windows expert. Any idea?

bernatx, 2020-07-06:
We recently found similar problems with boost and the last version of Python.
Try with Python 3.7.2 and be sure it is the x64 version.
Tell us if that works for any of you.

Thanks

closed 2020-09-12 by stale[bot]