carla-simulator/carla #2881: make PythonAPI boost python library link error windows 10 0.9.9

shaoxiaohang, 2020-05-24:
I was trying to run make PythonAP. It shows a bunch of boost python link error.
It says installation successful but no PythonAPI/dist is generated. 
make launch build fine and I can load the UE4 project.
I found the same issue reported before : https://github.com/carla-simulator/carla/issues/2339
But I couldn't find a solution for it.  Please help me on it. Just bought a new laptop for running Carla, its really frustrating. 
Carla version : 0.9.9
OS: Windows 10, AMD cpu.

closed 2020-05-24 by shaoxiaohang

ntutangyun, 2020-05-26:
hi @shaoxiaohang, have you managed to find the solution to this problem? thanks

shaoxiaohang, 2020-05-26:
Yes I was installing the 32 bit python. Make sure x64 python is installed
also run the x64 terminal in the vs tool not the regular CMB.

filippogandolfi, 2020-06-04:
> Yes I was installing the 32 bit python. Make sure x64 python is installed also run the x64 terminal in the vs tool not the regular CMB.
> […]

Sorry for the question, but what do you mean with "Make sure x64 python is installed"?
I have the same problem but I cannot adjust it

shaoxiaohang, 2020-06-04:
Check your python version, there are two versions of python,
32-bit/64-bit,  64-bit version is required.

filippogandolfi, 2020-06-07:
Thanks @shaoxiaohang, I had the correct type (64 bit) but for me works when I made this change:
- Uninstall python 3.8
- Install python 3.7 
- Remove anything related to boost under "carla/Build"
-  try make PythonAPI again
Thanks @marcgpuig for the help

Schuck84, 2020-07-11:
Hi @shaoxiaohang,
which Python x64 Version exactly did you use?
I'm using 3.7.7 (x64) and no other Python version is included in my PATH envvar, but I still have the same error and the PythonAPI isn't build (No /dist folder either .egg file is created)...
I made a make clean and removed the Build folder already
Thanks

Schuck84, 2020-07-11:
Hi all,
after uninstalling all my Python versions except 3.7.7 on my PC, the built is successfully finished.
In my PATH envvar I'm pointing to this version.
Is there another workaround than removing all Python versions?
It's bothering to remove all Python versions, as they are required for other programming stuff.
Are there other scripts searching the disk for Python versions?
Only changing the envvar was not effective.

I'm using Win10 + VS2017 + carla 0.9.9.4 from GIT clone...

shijieS, 2020-09-26:
I also met this problem. I find that there is a pre-installed python in Windows 10 (`C:/Windows/py.exe`). This python is used in `Carla/Util/BuildTools/BuildPythonAPI.bat` by calling `py ...` command. Therefore, here is my solution:
- replace all `py` in `Carla/Util/BuildTools/BuildPythonAPI.bat` by `python`
- replace `py -3 setup.py bdist_egg`  by `python setup.py bdist_egg` in line 112

By following these steps, I solved my problem.

I hope the official team can fix it in the next version.
I use CARLA0.9.9.4

Here is the final modified `Carla/Util/BuildTools/BuildPythonAPI.bat`
```python
@echo off
setlocal

rem BAT script that creates the client python api of LibCarla (carla.org).
rem Run it through a cmd with the x64 Visual C++ Toolset enabled.

set LOCAL_PATH=%~dp0
set "FILE_N=-[%~n0]:"

rem Print batch params (debug purpose)
echo %FILE_N% [Batch params]: %*

rem ============================================================================
rem -- Parse arguments ---------------------------------------------------------
rem ============================================================================

set DOC_STRING=Build and package CARLA Python API.
set "USAGE_STRING=Usage: %FILE_N% [-h^|--help] [--rebuild]  [--clean]"

set REMOVE_INTERMEDIATE=false
set BUILD_FOR_PYTHON2=false
set BUILD_FOR_PYTHON3=false

:arg-parse
if not "%1"=="" (
    if "%1"=="--rebuild" (
        set REMOVE_INTERMEDIATE=true
        rem We don't provide support for py2 right now
        set BUILD_FOR_PYTHON2=false
        set BUILD_FOR_PYTHON3=true
    )

    if "%1"=="--py2" (
        set BUILD_FOR_PYTHON2=true
    )

    if "%1"=="--py3" (
        set BUILD_FOR_PYTHON3=true
    )

    if "%1"=="--clean" (
        set REMOVE_INTERMEDIATE=true
    )

    if "%1"=="-h" (
        echo %DOC_STRING%
        echo %USAGE_STRING%
        GOTO :eof
    )

    if "%1"=="--help" (
        echo %DOC_STRING%
        echo %USAGE_STRING%
        GOTO :eof
    )

    shift
    goto :arg-parse
)

set PYTHON_LIB_PATH=%ROOT_PATH:/=\%PythonAPI\carla\

if %REMOVE_INTERMEDIATE% == false (
    if %BUILD_FOR_PYTHON3% == false (
        if %BUILD_FOR_PYTHON2% == false (
          echo Nothing selected to be done.
          goto :eof
        )
    )
)

if %REMOVE_INTERMEDIATE% == true (
    rem Remove directories
    for %%G in (
        "%PYTHON_LIB_PATH%build",
        "%PYTHON_LIB_PATH%dist",
        "%PYTHON_LIB_PATH%source\carla.egg-info"
    ) do (
        if exist %%G (
            echo %FILE_N% Cleaning %%G
            rmdir /s/q %%G
        )
    )
    if %BUILD_FOR_PYTHON3% == false (
        if %BUILD_FOR_PYTHON2% == false (
            goto good_exit
        )
    )
)

cd "%PYTHON_LIB_PATH%"
rem if exist "%PYTHON_LIB_PATH%dist" goto already_installed

rem ============================================================================
rem -- Check for python ------------------------------------------------------------
rem ============================================================================

where python 1>nul
if %errorlevel% neq 0 goto error_py

rem Build for Python 2
rem
if %BUILD_FOR_PYTHON2%==true (
    goto py2_not_supported
)

rem Build for Python 2
rem
if %BUILD_FOR_PYTHON3%==true (
    echo Building Python API for Python 3.
    python setup.py bdist_egg
    if %errorlevel% neq 0 goto error_build_egg
)

goto success

rem ============================================================================
rem -- Messages and Errors -----------------------------------------------------
rem ============================================================================

:success
    echo.
    if %BUILD_FOR_PYTHON3%==true echo %FILE_N% Carla lib for python has been successfully installed in "%PYTHON_LIB_PATH%dist"!
    goto good_exit

:already_installed
    echo.
    echo %FILE_N% [ERROR] Already installed in "%PYTHON_LIB_PATH%dist"
    goto good_exit

:py2_not_supported
    echo.
    echo %FILE_N% [ERROR] Python 2 is not currently suported in Windows.
    goto bad_exit

:error_py
    echo.
    echo %FILE_N% [ERROR] An error ocurred while executing the py.
    echo %FILE_N% [ERROR] Possible causes:
    echo %FILE_N% [ERROR]  - Make sure "py" is installed.
    echo %FILE_N% [ERROR]  - python = python launcher. This utility is bundled with Python installation but not installed by default.
    echo %FILE_N% [ERROR]  - Make sure it is available on your Windows "py".
    goto bad_exit

:error_build_egg
    echo.
    echo %FILE_N% [ERROR] An error occurred while building the egg file.
    goto bad_exit

:good_exit
    endlocal
    exit /b 0

:bad_exit
    endlocal
    exit /b %errorlevel%
    
```

Computer-CGuy, 2020-09-29:
Thanks a lot, @shijieS Your comment was helpful. 
Steps to do a successful build, the latest Carla commit 159801fd40b8de11a583fd63fb9bb45c42faff62 titled "Changelog" has errors on my build, for no clear reasons. So I downloaded the commit titled "New version 0.9.10". 
To download the commit
`
git init
git remote add origin https://github.com/carla-simulator/carla.git
git fetch origin 55a83f911a76852986fbb33a9d2560a4e2e0b284
git checkout FETCH_HEAD
`
And then follow the steps posted by @shijieS above.
Then make the repo as usual.

52bloo, 2022-02-18:
Hello. Seeing as the changes mentioned by [shijieS](https://github.com/carla-simulator/carla/issues/2881#issuecomment-699452386) were not implemented as of 0.9.13, I would like to further add that for those who'd like to have the scripts use `python` instead of `py` due to using conda on Windows, there are `py` calls in `BuildCarlaUE4.bat` and `Package.bat` to be modified as well. UE4 still seems to load without problems even if you don't change them, but it does print out a small complaint during `make launch` .