carla-simulator/carla #2882: make PyhtonAPI error on Windows.

AdelinaK, 2020-05-24:
Hello. I tried to build Carla on Windows but I've got this error on the command make PythonAPI. I searched a little bit and tried to install libpng on Windows from the official site, add it to the path but it still doesn't work. Even thought it says "successfully build" the C:\carla\PythonAPI\carla\dist folder was not created. Any ideas of what I can do?

C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.8\Release\source/libcarla/libcarla.obj /experimental:external /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
libcarla.cpp
dependencies/include/system\boost/gil/extension/io/png/tags.hpp(25): fatal error C1083: Cannot open include file: 'png.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\carla\PythonAPI\carla\dist"!

Avens666, 2020-05-25:
Do not install 3rd lib by yourself.
let Make command download and compile it in Build directory

AdelinaK, 2020-05-25:
It didn't work from the beginning so I tried to install it, but I've got the same error

Avens666, 2020-05-25:
1. check your Build directory had **libpng-1.2.37-install** directory
[image]

2. make PythonAPI will copy png.h file to  ./PythonAPI\carla\dependencies\include  directory
[image]

AdelinaK, 2020-05-26:
The files png.h and pngconf.h are missing in my include directory, maybe the make command could not copy them

germanros1987, 2020-06-26:
@bernatx could you please help with this?

AdelinaK, 2020-07-03:
Maybe it helps. I installed a x64 version of python 3.7.7 
I think the one I had before was x86 (32 bits version) and it does't work. So after the installation and make clean, make launch it worked for me.

bernatx, 2020-07-06:
Hi, that could be the case. We are facing some issues when users are using the x32 version of Python. Please be sure to use the x64 version of Python, and version 3.7.2 if possible, because last version of Python gives some linker problems with boost. We are still investigating.

closed 2020-09-12 by stale[bot]

hulaideyoushou, 2024-11-05:
try to delete the folder about"libpng-1.2.37-install"and"libpng-1.2.37-source",and **make PythonAPI** again

Reason1520, 2025-03-10:
> 尝试删除关于“libpng-1.2.37-install”和“libpng-1.2.37-source”的文件夹，并重新**制作 PythonAPI**

It works. Thank you for your solution.