carla-simulator/carla #2884: Logging data from scenario simulation

HarnarayanSingh, 2020-05-25:
Carla version: 0.9.7
OS- Ubuntu 18.04

Can we somehow log position, velocity, acceleration and other vehicle dynamics related data from the simulation to be later viewed in any other format?

HarnarayanSingh, 2020-06-09:
I have logged vehicle data for all possible vehicle actors using actor attributes, getters (get_location,...). Can we also do the same for pedestrian? There doesn't seem to be attributes for walker pedestrian to log data.

germanros1987, 2020-06-26:
@HarnarayanSingh why don't you just use the logger? https://carla.readthedocs.io/en/latest/tuto_G_retrieve_data/#start-recording

HarnarayanSingh, 2020-06-26:
@germanros1987 I had tried that. But i believe the document version only helps with key events of the simulation and not the whole data. So I logged data by modifying manual_control.py file. I'm still working on pedestrian data logging.

germanros1987, 2020-06-27:
Wouldn't it make more sense to derive all that information from the CARLA logs? In other words, to create a script that processes the log and derives the required information, e.g. velocity, acceleration, etc.

HarnarayanSingh, 2020-07-02:
@germanros1987 By Carla logs, do you mean the log file created on recording a simulation?

sergi-e, 2020-07-09:
@HarnarayanSingh I think he does. You can replay a simulation and retrieve the data for every frame recorded, instead of just some important ones. Just set the second parameter in the call for `show_recorder_file_info(self, filename, show_all=True)`. Maybe from there you cane derive the data you need.

HarnarayanSingh, 2020-07-09:
@sergi-e @germanros1987  Thanks for your help. Though I have been able to do the same by modifying manual_control.py script, I'll still try with the recorder option given that seems a good feature.  Will update and close this issue once i have tested that.

wpp201820601009, 2020-07-13:
At present, I have encountered similar problems. First of all, I want to use the trained neural network for lane change and Carla for simulation. The first step is to collect data, including the position, acceleration, speed of the vehicle, the turning angle of the vehicle relative to the road, the distance from the left (right) lane, whether the left (right) Lane exists, whether the left (right) Lane exists, and whether the vehicle is in front of, behind, left front and left rear The longitudinal distance of the right front and right rear vehicles, and then preprocess these data, input them into the trained model for lane change prediction, and collect data. If you use Carla log, is it not real-time data information, and then you can modify the manual_ control.py Script to get this information, there is no code to refer to, thank you very much, novice on the road, really a lot of things do not understand.

KunanonxGithub, 2020-09-16:
@wpp201820601009 So, if we want some data to train our neural network for making our cars run through some situations better, we need to write our code by ourselves because Carla log cant get us enough informations. Am I right?

tinmodeHuang, 2020-10-06:
@KunanonxGithub @wpp201820601009 hey! guys, did you achieve it? I have already read the documentation relevant to it [https://carla.readthedocs.io/en/latest/adv_traffic_manager/#simulation-state](url), from my point of view, we need to edit and  rebuild base code so that those states can be accessible

soheilade, 2020-10-13:
> 
> 
> Wouldn't it make more sense to derive all that information from the CARLA logs? In other words, to create a script that processes the log and derives the required information, e.g. velocity, acceleration, etc.

where are CARLA server logs stored?

wendersonj, 2020-10-23:
> > Wouldn't it make more sense to derive all that information from the CARLA logs? In other words, to create a script that processes the log and derives the required information, e.g. velocity, acceleration, etc.
> 
> where are CARLA server logs stored?

I just made a issue/topic answering that ! Check: https://github.com/carla-simulator/carla/issues/3504
In case that doesn't help you, check out other issues in carla's github, using the search tool. I recently saw others topics answering about that

wendersonj, 2020-10-23:
@germanros1987 maybe you can help me... 
I tried to use the function `client.start_recorder("/home/carla/recording01.log", True)` got this error:

`
Traceback (most recent call last):
  File "dqn_sync.py", line 637, in main
    client.start_recorder("logs/gravacao_ep_{0}.log".format(episodio), True)
Boost.Python.ArgumentError: Python argument types in
    Client.start_recorder(Client, str, bool)
did not match C++ signature:
    start_recorder(carla::client::Client {lvalue}, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name)
`
Is the function enabled to use `additional_data` argument ?

xinshuoweng, 2021-02-06:
> @HarnarayanSingh I think he does. You can replay a simulation and retrieve the data for every frame recorded, instead of just some important ones. Just set the second parameter in the call for [...]

Thanks for the explanation. I think it is possible to retrieve data like actor location, velocity, control signal, etc. But I am wondering is it possible to retrieve the sensor data (e.g., lidar and camera images) at every frame? Or in order to retrieve the same sensor data during replaying as the previously recorded one, should I add the same sensors to the same ego-vehicle during replaying?