carla-simulator/carla #2886: Help! I can not get full LiDAR points in one frame of data

Charlie-lusie, 2020-05-25:
As shown in figure below, I set the LiDAR sensor config correctly:
```
		sensors_bp['lidar']=[blueprint.find('sensor.lidar.ray_cast'),carla.Location(x=1.0,z=4.0)]
		sensors_bp['lidar'][0].set_attribute('rotation_frequency','10')
		sensors_bp['lidar'][0].set_attribute('channels','16')

```
But I can't get full points at once. It looks like data is generated in half of a full circle:
[image]

germanros1987, 2020-06-26:
@joel-mb @fpasch could this be a problem in our interface with rviz?

fpasch, 2020-06-26:
@Charlie-lusie Please ensure, that the rotation frequency equals the fixed_delta_seconds specified in carla_ros_bridge.

Charlie-lusie, 2020-06-27:
> @Charlie-lusie Please ensure, that the rotation frequency equals the fixed_delta_seconds specified in carla_ros_bridge.

I set the rotation frequency = 10Hz, and all params in carla_ros_bridge is default. But I fixed the problem by running: python config.py --fps=10

germanros1987, 2020-06-27:
@sergi-e before closing this issue, could you reflect this in the documentation, please?

PR #3005 (2020-07-01): sergi-e/issues_06
#### Description

Added some notes in the docs regarding issue solving: 
- A specific case for the Windows Build in FAQ.md
- A better explanation on how to tune the simulation when configuring the LIDAR to get for example, a full rotation per tick. 

Fixes #2886 #2642

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04.4
  * **Python version(s):** 3.6.9 and 2.7.17
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbacks

- More notes to come. 

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3005)
closed 2020-07-01 by sergi-e