carla-simulator/carla #2888: Actors created with wrong episode Id

BenAscent, 2020-05-25:
When spawning vehicles immediately after calling `load_world`, there is a chance that these vehicles got internally assigned the episode ID from the previous episode (I assume - or at least a wrong ID). The consequence of that is that their `IsAlive` function will always return `false`.
If these vehicles are controlled via the API (not autopilot/not registered in the traffic manager), they won't be properly registered for collision check (in LocalizationStage.cpp, there is a check for `IsAlive` for unregistered actors), and therefore will be ignored by other autopilot vehicles.
This was observed on one of our maps that is I believe much more heavy than the Town maps. We're not using the synchronous mode. Calling `wait_for_tick` after loading the map seems to solve this issue 100% (the issue itself was not occurring 100% of the time).
I believe adding a frame synchronization somewhere around `LoadEpisode` to ensure that subsequent calls to fetch the episode id return a correct value could fix the problem.

Env: Carla 0.9.9, Ubuntu 18, PythonAPI.2

glopezdiest, 2020-06-02:
@BenAscent Hello. Thanks for pointing that out. That is indeed a problem and we are currently working on it. We'll do a PR with the fix soon.

germanros1987, 2020-06-26:
Solved.

closed 2020-06-26 by germanros1987