carla-simulator/carla #2890: A Question about OpenDrive

lp249839965, 2020-05-26:
The RoadInfoLaneOffset element stores in Road according to 
https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/road/MapBuilder.cpp#L409
But it get from Lane in according to this code 
https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/road/Lane.cpp#L79

germanros1987, 2020-06-26:
Could you formulate your question and perhaps elaborate a bit more?

Axel1092, 2020-07-03:
Hi @lp249839965,
The RoadInfoLaneOffset is in a temporary container in https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/road/MapBuilder.cpp, which is then moved to the Lane object. Can you explain what problems are you having?

germanros1987, 2020-07-30:
@lp249839965 are you still interested in this issue?

closed 2020-08-29 by germanros1987