carla-simulator/carla #2892: Is it possible to use Carlar sensor model, Say RADAR in MATLAB driving scenario?

PradeepKumarBale, 2020-05-26:


bender2po, 2020-06-01:
@PradeepKumarBale I think, Matlab's Automated Driving Toolbox provides RADAR and LiDAR. Please see the link:
[https://in.mathworks.com/help/driving/getting-started-with-automated-driving-toolbox.html]
May I know why you want to use sensor implementation of Carla?
I am not much expert in this filed, but I think it is possible. I remember even Matlab uses UnReal engine for its 3D simulation. Since Carla sensor implementation is based on ray-tracing in UnReal you could use the module with some changes.

germanros1987, 2020-06-26:
@PradeepKumarBale long story short, we don't provide MATLAB bindings for the time being. Perhaps in the future, if there is enough interest.

closed 2020-06-26 by germanros1987