carla-simulator/carla #2897: Carla Traffic signs Information

aravindSwamy94, 2020-05-27:
Hi, 
      I suppose the traffic sign information in latest release can be obtained by 

map.get_all_landmarks()
.
I hope that this includes all the landmarks given in the 
carla.LandmarkType variables.

But when I run this .. The map.get_all_landmarks() returns an empty list. Am I doing some mistake for this API call? 
Please help me with this issue.

Thanks!

germanros1987, 2020-06-26:
@Axel1092 could you please comment on this? I guess landmarks are not registered yet in our old maps...

Axel1092, 2020-06-26:
Hi @aravindSwamy94, landmarks encode traffic signals defined in OpenDRIVE. However, our maps were created before the introduction of the landmark class and we are still working on combining the traffic signals that were already present in Carla. Therefore, landmarks are still not available in our maps but they are in the standalone mode, which is loading your own custom OpenDRIVE with traffic lights.

closed 2020-09-04 by Axel1092