carla-simulator/carla #2898: Failed to connect Carla server from a remote machine

StevePhan97, 2020-05-28:
Hi all,
I run the Carla server on a Windows PC and tries to connect a client running on a Linux remote machine.
Both machines are in the same network (using VPN). After running a server on port 2000, I can connect a client to `host_address:2000` via `telnet host_address 2000`. However, when I try a simple Carla script to get_world from the Carla client, I keep reporting time-out message: cannot connect to the server. Here is that script:
`
import carla
import settings
client=carla.Client(settings._HOST,settings._PORT)
client.set_timeout(10)
world = client.get_world()
`
One thing I notice, the host machine receives the message from the client during that 10 seconds, but the connection has not been successfully established:
[image].

FYI, the client's machine is actually an nvidia-docker container. I cannot set up the UnrealEngine on the client's machine since it couldn't display and render GUI. But the client machine only needs to connect to Carla environment on my host server and train the agent without rendering any images. It should be able to run PythonAPI without any problem, I assume.

Could you please advise?
Thank you 

Here are my system specifications:
Carla version: 0.9.8, note that I use a pre-built Carla by following "Quick installation" document,
Server's machine: Windows 10,
Client's machine: Ubuntu 16.04.

germanros1987, 2020-06-26:
@StevePhan97 how are you running the docker container? Please, provide more information.

StevePhan97, 2020-06-30:
@germanros1987 Hi, I have solved the issue. Thanks anyway.

closed 2020-06-30 by StevePhan97

charles-lei, 2020-07-23:
Hi @StevePhan101, I ‘m suffered the same issue, how did you solve this?

fabiankueppers, 2021-01-05:
Any solution to this?

wangxuan1997, 2021-11-18:
Any solution to this?