carla-simulator/carla #2899: Running docker without GPU support

umateusz, 2020-05-28:
Is it possible to run the simulator in a docker without GPU support, e.g. in no-renderring mode?

As I understand I can run a docker with the `--gpus all` flag and set no-rendering mode through PythonAPI. Is it possible to run on a machine without gpu access?

Armandomiguel, 2020-06-04:
Hello umateusz,

I wanted to do exactly the same and I ended up writing the Dockerfile from scratch and building the image in order to achieve this. 

You can try to the same (downloading from `http://dist.carla.org/carla-0.9.9/` and installing with `apt-get install carla-simulator`. I could also share my Dockerfile with you if you write me an email to armando.miguel.garcia[at]gmail.com; because of an ongoing project, I'm not able to publish it here.

But that what you want to do is definitely possible!
Best regards

umateusz, 2020-06-21:
Sorry for the late response. Thank you, you encouraged me to create such adockerfile. The simulator is currently exiting without writing an error if the `--gpus=all` flag is not used.

I think this is a temporary solution. I think that such a solution would be very useful. I am not sure how many changes it requires to make it possible.

germanros1987, 2020-06-26:
@umateusz chances are that the simulation won't work without a GPU, even in non_rendering_mode. If you achieve it, please report back.

hh0rva1h, 2020-07-20:
@germanros1987 Unfortunately it seems not possible at the moment. We have access to a HPC cluster where Carla refuses to start due to a GPU issue:

```
[2020.06.25-16.35.56:623][  0]LogInit: Using SDL_WINDOW_OPENGL
[2020.06.25-16.35.56:632][  0]LogLinux: Warning: MessageBox: Warning: OpenGL is deprecated, please use Vulkan.: Message: 
[2020.06.25-16.35.56:640][  0]LogLinux: Warning: MessageBox: Cannot create OpenGL-enabled SDL window. SDL error: 'Cannot create an OPENGL window invalid egl_data'.: Cannot create SDL window.: 
[2020.06.25-16.35.56:640][  0]LogCore: FUnixPlatformMisc::RequestExit(1)
```
Starting in norendering mode leads to the same error. Also starting on a server without GPU in no rendering mode fails.
Would be really great if someone could work on making it possible to work without GPU, but maybe this is not possible due to PhysX?

germanros1987, 2020-07-30:
@hh0rva1h we will make this a feature request. Hopefully, we will be able to take care of this in future releases.

sohamm17, 2020-09-27:
Is it possible to run Carla with Intel GPUs? If yes, do you know the command to run the docker image?

germanros1987, 2020-09-27:
Do you mean Intel integrated GPUs? I don't know if docker has a specific plugging to support integrated GPUs. This is not very conventional.

sohamm17, 2020-09-27:
> Do you mean Intel integrated GPUs? I don't know how mature is that. You can take a look at this : https://hub.docker.com/r/intel/intel-gpu-plugin

Thanks. Yes, I tried it. Did not work well.
Does Carla run in the native form, not the docker image, without Nvidia GPU? Just Intel CPU and Integrated GPUs.
I tried, but it was failing with segmentation fault, with `opengl`. With `vulkan`, it ran and showed the window but the computer got hang up. So I was wondering if anyone has tried with just Intel GPUs and what the experience is.

closed 2020-12-25 by stale[bot]

qhaas, 2021-02-11:
> @umateusz chances are that the simulation won't work without a GPU, even in non_rendering_mode. If you achieve it, please report back.

I am aware of a team at a major institution that spent quite a bit of time trying to get CARLA/UnrealEngine working without GPUs (i.e. CPU only with dual-socket Xeon servers) and they did not succeed.

Armandomiguel, 2021-02-11:
@qhaas 
@germanros1987  
It is indeed possible to run CARLA without a dedicated GPU, e.g. Nvidia.

I wrote two versions of a Dockerfile for a container with the simulation, which can perfectly run on Intel Mesa:
The first version downloads, installs and executes CARLA. On a simple computer this takes relatively long for the building (>25 min.) and the resulting image has over 9.5GB, but it works.
The second version mounts a packaged CARLA, which improves the building time and resulting image size by a great amount. This is my preferred way of deploying CARLA.
I could share those versions directly with you, just drop me an email at armando.miguel.garcia[at]gmail.com.

Best regards

sohamm17, 2021-02-12:
It did ultimately work out for me with Intel's iGPU, although it is very slow. I have an Intel Coffee Lake processor.

SimonLeeGit, 2021-04-28:
> @umateusz chances are that the simulation won't work without a GPU, even in non_rendering_mode. If you achieve it, please report back.

hi, I'm doing the same thing right now. And I have found a way to achieve this. If I run carla without camera sensor support, I can use the dedicated server mode, and it can run with a no rendering mode other than offline rendering. It's useful when I do the simulation for traffic world.

ShahudullahKhan, 2021-08-28:
> It did ultimately work out for me with Intel's iGPU, although it is very slow. I have an Intel Coffee Lake processor.

Hi @sohamm17 , Can you please share the solution . I am also trying to install Carla in  a similar system with Mesa Intel® HD Graphics 5500 (BDW GT2) and without Nvidia GPU. 

Thanks,

markomiz, 2023-04-26:
Hi, I have just come across this thread looking for a solution to a similar issue. I wanted to run Carla on Ubuntu 22.04 (other parts of my project require this) but this isn't yet supported - so I then thought I could run it in docker instead, however my machine does not have a gpu, nor an intel processor (AMD Ryzen) - For those who have found solutions for the docker image without gpu - are you able to give some suggestions on ways to get this working? (if at all possible)