carla-simulator/carla #2905: what is the form of Lidar measurements raw data

Deio828, 2020-06-01:
Hi, 
I'm using Carla0.9.6 and I'm trying to use the lidar point cloud, I've tried to use `save_to_disk()` method as:
`lidar.listen(lambda data : data.save_to_disk('file_name.ply'))`
I got the first 10 points as shown here:
`9.2588 -22.5821 -4.4486`
`8.3601 -19.6354 -3.8891`
`7.8168 -17.6970 -3.5250`
`8.4143 -18.3785 -3.6822`
`8.4126 -17.7419 -3.5762`
`8.4109 -17.1406 -3.4767`
`8.4094 -16.5714 -3.3832`
`8.4079 -16.0319 -3.2950`
`8.4065 -15.5194 -3.2119`
`8.4052 -15.0318 -3.1333`

but when I try to use these readings as in this function:

```
def do_something(data):
	out = data.raw_data
	out = np.array(out)
	out = out.reshape((int(len(out)/3.0) , 3))
```

and call it like this:

`lidar.listen(lambda data: do_something(data))`

I get the first 10 points to be:
`[247  35  20]`
`[ 65  62 168]`
`[180 193 254]`
`[ 90 142 192]`
`[198 194   5]`
`[65 82 21]`
`[157 193 190]`
`[231 120 192]`
`[117  35 250]`
`[ 64 107 147]`
`

please, how can I make the 10 points generated by `do_something` function to be like the 10 points generated in the `save_to_disk` method?
thanks in advance.

jldez, 2020-06-03:
out = np.frombuffer(data.raw_data, dtype=np.dtype('f4'))
out = np.reshape(out, (int(out.shape[0] / 3), 3))

Deio828, 2020-06-04:
@jldez , Yes that worked properly, thanks alot

closed 2020-06-26 by germanros1987

leishi23, 2022-07-06:
But what's the difference between numpy.frombuffer & numpy.array here? The documentation says both can create an array. Thank you in advance!