carla-simulator/carla #2906: Runtime error in carla 0.9.5 and 0.9.4 on windows 10

xusafzai, 2020-06-02:
Hi, i'm getting these errors alternatively while running my program in Carla.
either I get this error:
"RuntimeError: failed to connect to newly created map"
or this:
"RuntimeError: time-out of 20000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000"
Also I have both versions 0.9.5 and 0.9.4 but in version 0.9.4 i'm can't find any file named "Carla" and "EGG INFO" in python API.
i'm attaching the code If someone wants to have a look.

[code.txt](https://github.com/carla-simulator/carla/files/4713997/code.txt)

germanros1987, 2020-06-26:
Have you tried with CARLA 0.9.9? Does the problem persist?

germanros1987, 2020-07-30:
Issue closed due to lack of activity. Feel free to open a new issue if you still consider this is a critical matter.

closed 2020-07-30 by germanros1987