carla-simulator/carla #2909: Running Carla Server and Client in a Docker

Armandomiguel, 2020-06-03:
Hello everyone,

Has anyone experience at running a Carla-client in a Docker?
I'd like to have both the server and the client running in containers. I'm aware of the tutorial for the server-side but I haven't found anything about the client-side.

Could you please help me?
Thanks

Armandomiguel, 2020-06-04:
Hello again,

Update: I already have both the client and the server running, but the communication through the ports (2000-2002) is not working properly.

I'm running the server-container using `<... -p 2000-2002:2000-2002 ...>`. And the communication between this server and my local computer works fine (i.e. container-host communication instead of container-container), but when trying to assign the same ports to the client, I received the following error message:
`Error starting userland proxy: listen tcp 0.0.0.0:2002: bind: address already in use.` 

So, does somebody know how to connect both the server and client containers?

Any help is appreciated.
Thanks

benderdave, 2020-06-04:
We've been able to run both the server and client inside docker containers without any issues. You should not need to map ports on the client docker container, since the client connects to those ports as a client (rather than binding to them as a server). Just use '--net=host' on the client container and it should work.

Armandomiguel, 2020-06-04:
Thank you very much benderdave! 

"--net=host" on the client side is exactly what I needed. Now they run both on containers and communicate with each other. 

Thanks again!

closed 2020-06-04 by Armandomiguel