carla-simulator/carla #2910: self.world.tick() error on synchronous.py example with Unreal Editor (CARLA 0.9.6)

MartinDubreuil, 2020-06-03:
After running the synchronous.py example with UnrealEditor running, if I made a simple left-click in the "Content Browser" section to navigate inside Carla Content, the script crashes with this error message :

```
Traceback (most recent call last):
  File "synchronous_mode.py", line 204, in <module>
    main()
  File "synchronous_mode.py", line 170, in main
    snapshot, image_rgb, image_semseg = sync_mode.tick(timeout=2.0)
  File "synchronous_mode.py", line 78, in tick
    self.frame = self.world.tick()
RuntimeError: frame == _episode->GetState()->GetTimestamp().frame
```

Is there a way to fix this bug ?

Thanks !

germanros1987, 2020-06-26:
@MartinDubreuil could you please provide more (much more) information about your environment, version of CARLA, etc?

@bernatx please be ready to look into this.

bernatx, 2020-07-06:
@MartinDubreuil this is a normal behaviour. 
The problem is that when you make any interaction in editor that makes it busy, the client part (your script) can not communicate well with the server (editor) and if it takes more than 2 seconds (see your default time out when you call the tick) to get an answer, the script terminates thinking that the server does not exist anymore.
You can use a greater timeout when calling the tick(timeout).
I hope that resolves the problem.

closed 2020-09-12 by stale[bot]