carla-simulator/carla #2914: unable to parse the OpenDrive XML string

xli4217, 2020-06-03:
Getting error "unable to parse the OpenDrive XML string" when running scenarios in scenario_runner

- Steps followed
1. Start Carla ./CarlaUE4./sh
2. Run scenario `python scenario_runner.py --scenario OtherLeadingVehicle_1 --reloadWorld`
Expected behavior

- System
OS: [Ubuntu 18.04]
CARLA Version [CARLA 0.9.9.2]
Python version [3.7]
Scenario_runner version [0.9.9]

whuxhzhang, 2020-06-04:
I have the same problem as you

fabianoboril, 2020-06-26:
@jackbart94 I think this was fixed with 0.9.9.4, or not?

jackbart94, 2020-06-26:
Hi,

As far as I know, according to what we discussed in [this issue](https://github.com/carla-simulator/scenario_runner/issues/556) the problem is fixed with 0.9.9.4. Please try with this version of CARLA and let us know!

Alikerin, 2020-07-13:
Hi everyone.. I need some help.

I setup CARLA on a Google VM and I have tried setting up the scenario_runner as well but I have been met with different challenges. I am able to run the PythonAPI examples of CARLA but not the manual_control.py of scenario_runner. It keeps saying "Waiting for the ego vehicle..." or "Scenario Not Ready". When I also tried running `python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld` it returns "Not all tests were successful". I have tried using (same versions for carla and scenario runner) v0.9.8 and 0.9.9. The error I get on my server is "unable to parse the OpenDrive XML string"

Alikerin, 2020-07-15:
> Hi everyone.. I need some help.
>
> I setup CARLA on a Google VM and I have tried setting up the scenario_runner as well but I have been met with different challenges. I am able to run the PythonAPI [...]

@jackbart94 @fabianoboril I have been able to fix this issue with `manual_control.py` by spawning a hero vehicle before passing the world object to the WorldSR class in `game_loop` [here](https://github.com/carla-simulator/scenario_runner/blob/master/manual_control.py). I will like to know the originally intended behavior. Thanks

closed 2020-09-20 by stale[bot]