carla-simulator/carla #2915: Computing relative transformations between objects / How does carla.Transform.transform() work

JWHennessey, 2020-06-03:
Hi, 

I'm trying to compute the relative transformation between two actors in Carla Actor A and Actor B. 

To do this I take the two carla.Transform objects and convert them into numpy 4x4 transformation matrices. 

I compute the relative transform (T_a2b) from A to B then convert the 4x4 matrix back into a Carla transform. 

To verify that this works I want to validate that `B' = T_a2b * A`  and `B' == B`. 

If I do this using numpy as 4x4 matrix then this makes sense. However, applying the transformation using `carla.Transform.transform` doesn't result in the same behaviour. 

Can someone explain how `carla.Transform.transform` works? Or suggest an alternative way to compute the relative transform between two actors. Thanks!

```
def carla_vec2numpy(vec):
    return np.array([vec.x, vec.y, vec.z])

def transform2mat(t):
    xyz = carla_vec2numpy(t.location)
    pitch = t.rotation.pitch
    roll = t.rotation.roll
    yaw = t.rotation.yaw
    R = transforms3d.taitbryan.euler2mat(
        np.deg2rad(pitch), np.deg2rad(roll), np.deg2rad(yaw)
    )

    M = np.zeros((4, 4))
    M[0:3, 0:3] = R
    M[0:3, 3] = xyz
    M[3, 3] = 1

    return M

def mat2transform(M):

    pitch, roll, yaw = transforms3d.taitbryan.mat2euler(M[0:3, 0:3])
    roll = np.rad2deg(roll)
    pitch = np.rad2deg(pitch)
    yaw = np.rad2deg(yaw)

    T = carla.Transform(
        carla.Location(x=M[0, 3], y=M[1, 3], z=M[2, 3]),
        carla.Rotation(pitch=pitch, yaw=yaw, roll=roll),
    )

    return T

def relative_transform(source, target):
    source_t = transform2mat(source)
    target_t = transform2mat(target)
    source_t_inv = np.linalg.inv(source_t)

    relative_transform_mat = np.dot(target_t, source_t_inv)

    relative_transform_transform = mat2transform(relative_transform_mat)

    return relative_transform_transform

def main():

    #Input transform A
    T_a = carla.Transform(
        carla.Location(2, -5, 4), carla.Rotation(pitch=0, yaw=-45, roll=0)
    )

    #Input transform B
    T_b = carla.Transform(
        carla.Location(-1, 6, 2), carla.Rotation(pitch=10, yaw=45, roll=0)
    )

    #Compute relative transform
    T_a2b = relative_transform(T_a, T_b)

    #Convert to a 4x4 matrix
    T_a2b_mat = transform2mat(T_a2b)

    #Transform T_a to T_b using numpy
    T_b_prime_np = np.dot(
        T_a2b_mat, np.array([T_a.location.x, T_a.location.y, T_a.location.z, 1]),
    )

    #Transform T_a to T_b using carla 
    T_b_prime = T_a2b.transform(T_a.location)

    #Expect both outputs to be -1, 6, 2
    print("T_b_prime_np", T_b_prime_np) # [-0.99999987  5.99999998  1.99999998  1.] - ~Correct 
    print("T_b_prime ", T_b_prime) # Vector3D(x=0.407027, y=11.069320, z=11.481960) - Incorrect

if __name__ == "__main__":
    main()
```

germanros1987, 2020-06-26:
@DSantosO it may be interesting for you to respond this question.

DSantosO, 2020-06-30:
Hello @JWHennessey,

Be careful because I think the transforms3d.taitbryan does not use the same convention for the angles that Unreal and us are using. Also, I think that when you are computing B' = T_a2b * A, you are not taking in consideration the rotation in A.

Anyway, in the PR #2997 (soon in master), we have exposed the 4-matrix form of the transformation so you can now easily use them to compute the relative transformation from A to B. To do that you just need to :
TaI = np.array(T_a.get_inverse_matrix())
Tb =  np.array(T_b.get_matrix())
Tab = np.dot(dso_Tb, dso_TaI)

I hope this is useful.

closed 2020-09-05 by stale[bot]

kannidekan, 2020-12-13:
First, there is a error in your function `relative_transform`.
If you wanna transform from **source** coordinate to **target** one. It should be below.
`def relative_transform(source, target):
    source_t = transform2mat(source)
    target_t = transform2mat(target)
    target_inv = np.linalg.inv(target_t)

    relative_transform_mat = np.dot(target_inv , source_t)

    relative_transform_transform = mat2transform(relative_transform_mat)

    return relative_transform_transform`
Because, carla.Transform.transform works like[ this](https://carla.readthedocs.io/en/latest/python_api/#carlatransform:~:text=Translates%20a%203D%20point%20from%20local,current%20transformation%20as%20frame%20of%20reference.):
> Translates a 3D point from local to global coordinates using the current transformation as frame of reference.

Second, transform in Carla maybe is not as same as the transforms3d.taitbryan. Here is the #553 .
[image]

Finally, i code the transform following below:
`def transform2M(transform):
    """generate transform matrix with carla.transform in Carla.

    Parameters
    -------------
    transform: carla.Transform
        tansform in carla.

    Return
    -------------
    M: np.ndarray
        transform matrix
    """
    location, rotation = transform.location, transform.rotation
    x, y, z = location.x, location.y, location.z
    pitch, yaw, roll = radians(rotation.pitch), radians(rotation.yaw), radians(rotation.roll)

    Rz = np.array([[cos(yaw), -sin(yaw), 0], [sin(yaw), cos(yaw), 0], [0, 0, 1]])
    Ry = np.array([[cos(pitch), 0, -sin(pitch)], [0, 1, 0], [sin(pitch), 0, cos(pitch)]])
    Rx = np.array([[1, 0, 0], [0, cos(roll), sin(roll)], [0, -sin(roll), cos(roll)]])
    R = Rz.dot(Ry).dot(Rx)
    t = np.array([[x], [y], [z]])

    M = np.hstack((R, t))
    M = np.vstack((M, np.array([[0,0,0,1]])))
    return M
`
You can test with this.
`_transform = carla.Transform()
_transform.location.x = 1
_transform.location.y = 2
_transform.location.z = 3

A = _transform.get_matrix()
A = np.array(A)
B = transform2M(_transform)
if np.array_equal(A, B):
        print('Equal.')
else:
        print('Not equal.')
        print('A: ')
        print(A)
        print('B: ')
        print(B)`
Hope it help you.