carla-simulator/carla #2917: run carla0.9.9 error in ubuntu18.04

whuxhzhang, 2020-06-04:
when I launch CarlaUE4.sh, it will report this error:

4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=127232 LargeMemoryPoolOffset=258368 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

zkyxsq, 2020-06-04:
I have same question with you

whuxhzhang, 2020-06-04:
> I have same question with you

I used my own carla_map, deleted some materials, recompiled it, and temporarily solved this problem.
But I don't think it's a good way.

guyu2711, 2020-06-08:
I have the same error with my laptop (940m), but it works fine on my PC with GTX 970. So it might need a better graphics card.

germanros1987, 2020-06-26:
@whuxhzhang do you care to provide more details, or should we just close this issue?

zhoushiyang12, 2020-07-01:
> > I have same question with you
> 
> I used my own carla_map, deleted some materials, recompiled it, and temporarily solved this problem.
> But I don't think it's a good way.

I deleted some materials too, in UE editor. But, how to recompiled it, and apply it to the Carla-Release?

whuxhzhang, 2020-07-01:
> > > I have same question with you
> > 
> > 
> > I used my own carla_map, deleted some materials, recompiled it, and temporarily solved this problem.
> > But I don't think it's a good way.
> 
> I deleted some materials too, in UE editor. But, how to recompiled it, and apply it to the Carla-Release?

click "save" button and "build" button in UE4.  Besides, you should download carla source code, then make package.

zhoushiyang12, 2020-07-01:
I think i have build

> click "save" button and "build" button in UE4. Besides, you should download carla source code, then make package.

Hello,  thank you for your help. I think i have buildt  it and i can run it in UE_edit .But where can i find the  compiled  files,  in which can i apply it to the Carla-Release?

whuxhzhang, 2020-07-01:
> I think i have build
>
>
> Hello, thank you for your help. I think i have buildt it and i can run it in UE_edit .But where can i find the compiled files, in which can i apply it to the Carla-Release?

file locate in carla/Unreal/CarlaUE4/Saved/Cooked/LinuxNoEditor/CarlaUE4/Content/Carla/Maps, end with .uexp

zhoushiyang12, 2020-07-01:
> file locate in carla/Unreal/CarlaUE4/Saved/Cooked/LinuxNoEditor/CarlaUE4/Content/Carla/Maps, end with .uexp

Oh! That does work! Thank you so much!

closed 2020-07-02 by whuxhzhang