carla-simulator/carla #2918: OpenGL

Yiquan-lol, 2020-06-04:
Hello, I installed Carla on the Linux system by "Quick install", but when running "./ carlaue4.sh ", I was prompted "Can not create OpenGL context.Check out that the drivers and Hardware support at least OpenGL 4.3".What should I do, please

zerone1996, 2020-06-11:
Hello, have you solved that? I have met the same problem.

Yiquan-lol, 2020-06-11:
> 你好，你解决了吗？我遇到了同样的问题。

Sorry, I didn't solve it. This seems to be a common problem with Linux.

Yiquan-lol, 2020-06-11:
> Hello, have you solved that? I have met the same problem.

Sorry, I didn't solve it. This seems to be a common problem with Linux.

germanros1987, 2020-06-26:
@Yiquan-lol could you please provide us with more information about your system? For instance...version of CARLA, GPU that you are using, driver, etc.

Yiquan-lol, 2020-06-26:
I've solved my problem.I plugged the display into the Nvidia graphics card and updated the Nvidia driver to the latest version.

Yiquan-lol, 2020-06-26:
@zerone1996  You need to update the Nvidia driver to the latest version and have the display attached to the Nvidia graphics card

closed 2020-06-27 by germanros1987

carlaUE4new, 2020-08-04:
> @zerone1996 You need to update the Nvidia driver to the latest version and have the display attached to the Nvidia graphics card

> @zerone1996 You need to update the Nvidia driver to the latest version and have the display attached to the Nvidia graphics card

@Yiquan-lol , how do you have the display attached to the Nvidia graphics card?