carla-simulator/carla #2922: Failure to compile PythonAPI in Windows 10

ZhengyaoYu, 2020-06-04:
I am able to compile the server simulator through make launch. However, I got the following error messages when make PythonAPI. I saw a few similar issues but could not find a solution. Any suggestion on how to fix this would be greatly appreciated. 

OS: Windows 10 1809
CARLA version: 0.9.9
Python 3.8.3 (64 bit)

TrafficManagerRemote.cpp
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:dependencies/lib /LIBPATH:C:\Users\yzh\AppData\Local\Programs\Python\Python38\libs /LIBPATH:C:\Users\yzh\AppData\Local\Programs\Python\Python38\PCbuild\amd64 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\um\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\um\x64" /EXPORT:PyInit_libcarla build\temp.win-amd64-3.8\Release\source/libcarla/libcarla.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\Buffer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\Exception.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\FileSystem.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\StringUtil.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\Actor.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\ActorAttribute.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\ActorBlueprint.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\ActorList.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\BlueprintLibrary.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\DebugHelper.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\Junction.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\LaneInvasionSensor.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\Light.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\LightManager.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\Map.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\ServerSideSensor.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\TimeoutException.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\TrafficLight.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\TrafficSign.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\Vehicle.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\Walker.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\WalkerAIController.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\Waypoint.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\World.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\detail\ActorFactory.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\detail\ActorState.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\detail\ActorVariant.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\detail\Client.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\detail\Episode.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\detail\EpisodeProxy.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\detail\EpisodeState.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\detail\Simulator.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\client\detail\WalkerNavigation.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\geom\GeoLocation.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\geom\Math.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\geom\Mesh.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\geom\MeshFactory.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\nav\Navigation.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\nav\WalkerEvent.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\nav\WalkerManager.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\opendrive\OpenDriveParser.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\opendrive\parser\ControllerParser.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\opendrive\parser\GeometryParser.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\opendrive\parser\GeoReferenceParser.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\opendrive\parser\JunctionParser.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\opendrive\parser\LaneParser.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\opendrive\parser\ObjectParser.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\opendrive\parser\ProfilesParser.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\opendrive\parser\RoadParser.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\opendrive\parser\SignalParser.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\opendrive\parser\TrafficGroupParser.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\pointcloud\PointCloudIO.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\road\Lane.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\road\LaneSection.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\road\Map.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\road\MapBuilder.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\road\MapData.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\road\MeshFactory.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\road\Road.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\road\SignalType.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\road\element\Geometry.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\road\element\LaneCrossingCalculator.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\road\element\LaneMarking.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\road\element\Waypoint.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\rpc\WeatherParameters.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\Deserializer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\data\LaneInvasionEvent.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\s11n\CollisionEventSerializer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\s11n\DVSEventArraySerializer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\s11n\EpisodeStateSerializer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\s11n\GnssSerializer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\s11n\ImageSerializer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\s11n\IMUSerializer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\s11n\LidarSerializer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\s11n\NoopSerializer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\s11n\ObstacleDetectionEventSerializer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\s11n\RadarSerializer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\sensor\s11n\SensorHeaderSerializer.obj build\temp.win-amd64-3.8\Release\dependencies/include/carla\streaming\detail\Dispatcher.o [... cut ...]
libcarla.obj : error LNK2001: unresolved external symbol "class boost::python::api::object __cdecl boost::python::api::getitem(class boost::python::api::object const &,class boost::python::api::object const &)" (?getitem@api@python@boost@@YA?AVobject@123@AEBV4123@0@Z)
build\lib.win-amd64-3.8\carla\libcarla.cp38-win_amd64.pyd : fatal error LNK1120: 76 unresolved externals
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\Users\yzh\Documents\smart_city\carla\PythonAPI\carla\dist"!

germanros1987, 2020-06-26:
@bernatx @sergi-e could you help with this?

bernatx, 2020-07-06:
Hi,
Please be sure that you are using the x64 version of Python, and use Python 3.7.2, as it seems the last version is giving this kind of errors.
We are still investigating, but probably this will fix your problem.
Let us know please.
Thanks

ZhengyaoYu, 2020-07-13:
@bernatx Thank you for the suggestion. I just tried Python 3.7.2 and the issue persists.

closed 2020-09-20 by stale[bot]