carla-simulator/carla #2923: CARLA with CarSim Plugin

jackycstw, 2020-06-05:
I am trying to put CarSim unreal plugin into CARLA. But there are some questions.
1. CarSim work as CarSimPawn class in unreal engine. But CARLA use carla.carlawheeledvehicle class in unreal engine. I think there are two way to approach this:
a. Try to make a CarSimVehicleFactory, put CarSimPawn into it and spawn it from python client.
The difficulty is I don't know how to build a custom blueprint factory. Is there any tutorials about that?
b. Try to make CarSimPawn inherit from carlawheeledvehicle.
I have no idea about how to do this. CarSimPawn has it's own movement component, cannot be replaced easily from carlawheeledvehicle.

Please give me some ideas about how to solve this problem.

germanros1987, 2020-06-26:
@jackycstw we are currently trying to integrate CarSim with CARLA. We don't have good answers to your questions yet (we are not using the UE4 plugin). Once our integration is further developed we will happily provide more details.

jackycstw, 2020-06-29:
@germanros1987 
Thanks for your help.
Actually, I solved this problem myself last week. The key point is make CarSim's pawn inherit from Carla's class.

germanros1987, 2020-06-29:
Would you care to share more details with us?

jackycstw, 2020-06-30:
@germanros1987  I have projects on this issue so I can't share it in this moment. I think I can share it after the project finished.

dineshresearch, 2020-08-25:
Any update on this. I am curious to know how @jackycstw has do the integration of carla and carsim both

closed 2021-04-27 by stale[bot]