carla-simulator/carla #2925: Problem with adding new car

villimikko, 2020-06-05:
Hello

I am having issues with adding new vehicle to Carla. I have downloaded genral4wheelskeleton and binded the bones (I think) to mesh with blender. Picture 1.  But when im trying to do import to UE4 carla,I cant find those bones picture 2.

What am I doing wrong?

[image]

[image]

germanros1987, 2020-06-26:
@XGodina @gsmario could you help with this?

gsmario, 2020-06-26:
Hello @villimikko 
First, sorry for the delay, let me see if I can help you.
At first glance, I think that there was a problem exporting the model and the skeleton. 
You need to bind the bones to each different part of the car, and then export everything into an fbx. Please, make sure you have selected both the skeleton and the meshes. 
If that doesn't solve your problem, please take a screenshot of the software you are using to bind the bones, so I can check if there is another problem.
Thanks for  your time!

villimikko, 2020-06-29:
**
[image]
**

Here is my binding in blender.

gsmario, 2020-06-29:
Hello @villimikko 
Sadly, we don't use blender, so I'm not able to guide you through this process step by step. 
I recommend you using autodesk maya (you can use a free student license) to bind the skeleton, this way, I'll be able to help you.
I'll try to bind the skeleton with blender as soon as I can, but as I said, I'm not a blender user, so I'll need some time.

Thank you again.

germanros1987, 2020-06-29:
@gsmario I think this is a good opportunity for us. We should make the effort to create a basic vehicle in Blender in order to advise future users on how to do this.

@villimikko could you please provide the sources?

maxime-derache, 2020-06-30:
Please find below a tutorial that I made on this subject. I hope it will help you to fix your issue!

## Create the armature
* **Shif+A** [Add/Armature]

[image]

* ViewportDisplay Shapes=yes, Infront=yes, Axes=yes, GroupColors=yes
* Rotation X=90
* Scale X=100, Y=100, Z=100

[image]

## Create the bones
* Select body (in Object Mode)  
* **Shift+S** (CSursor to selected)
* Select the **Armature**
* Right-clic [Snap/Selection to cursor]
* **Tab** (switch to **Edit Mode**)
* **E** (extrude)
* **Y** (Y-axis)
* **Alt+P** [Disconnect Bone]

[image]

* **G** (move)
* **X** (X-Axis)
* **Shift+D** (duplicate)
* **X** (X-Axis)
* Repeat the last 2 actions 3 times 

[image]

## Set position of bones to rotation origin of the wheels
* **Tab** (switch to **Edit Mode**)
* Repeat for each wheel: 
    * Set the cursor position to the wheel center
    * Select a bone for wheel
    * Right-clic [Snap/Selection to cursor]

[image]

* **Tab** (switch to **Object Mode**)
* **A** (select all)
* **Ctrl+A** [Apply/Rotation & scale]

[image]

## Join the armature to the meshes
* Select each object and set its name:
    * "Wheel_Front_Right"
    * "Wheel_Front_Left"
    * "Wheel_Rear_Right"
    * "Wheel_Rear_Left"
    * "Body"
    * ...
* Ctrl-select all objects in the list, with the armature at the end

[image]

* **Ctrl+P** [Set parent to/Armature deform/With empty groups]
* Check that all meshes are under the armature in the tree view

[image]

## Associate meshes to corresponding bones
* **Tab** (switch to **Edit Mode**)
* Select each bone and set its name:
    * "Root"
        * "Wheel_Front_Right"
        * "Wheel_Front_Left"
        * "Wheel_Rear_Right"
        * "Wheel_Rear_Left"

[image]

* Associate each meshes to the corresponding bone:
    * **Tab** (switch to **Object Mode**)
    * Select an object
    * **Tab** (switch to **Edit Mode**)
    * Check that whole object meshes are well selected
    * Select the vertex group corresponding to the mesh
    * Press **Assign** button

[image]

## Check the armature definition
* Select Armature/Pose to switch to **Pose Mode**
* Check each mesh association:
    * Select the bone
    * **R** (Rotate)
    * **Echap**

[image]

* Check the good axes orientation of the bones (X Forward, Z Up)

[image]

## Export the FBX file with dynamic mesh and skeleton
* Menu [File/Export/FBX (.fbx)]
* Include/Object Types="Armature" and "Mesh" (shift-select)
* Transform/Forward="X Forward"
* Transform/Up="Z Up"
* Geometry/Smoothing="Face"
* Armature/Add Leaf Bones=*NO*
* Brake Animation=*NO*

[image]

maxime-derache, 2020-06-30:
## Create the vehicle Blueprints folder in Carla project
* Open the Acvsare Unreal project simulation/avcsare/Unreal/CarlaUE4/CarlaUE4.uproject
* Create a new folder with the name of the vehicle "*MyVehicle*" in Content/Carla/Blueprints/Vehicles

## Import the mesh and skeleton in Carla project
* Press **Import** button and open the FBX file that contain the meshes and skeleton of the vehicle
* Check that **Skeleton** and **Mesh** are selected
* **Import all**
* The mesh and skeleton are displayed in the assets panel

[image]

## Configure the PhysicsAsset
* Open *MyVehicle*_PhysicsAsset
* Select the wheels in the **Skeleton Tree**

[image]

* **Suppr** (Remove the collision bodies)
* Thumbnail [Tools/Body Creation] Primitive Type=Sphere
* Thumbnail [Tools/Body Creation] Auto Oriented to Bone=*NO*
* **Re-generate Bodies**

[image]

* **Save**
* Close the asset editor

## Add VehicleWheel Blueprint
* **Add New** Blueprint Class
* [Pick Parent Class/All Classes] Search="VehicleWheel"
* **Select**
* Rename the asset "*MyVehicle*_Wheel"

[image]

* Open the asset
* **Class Defaults**
* [Details/Wheel] Shape Radius=#Radius measured in Blender#
* [Details/Wheel] Shape Width=#Width measured in Blender#
* [Detail/Tire] Tire Config="CommonTireConfig"

[image]

* **Compile**
* **Save**
* Close the asset editor

## Add Animation Blueprint
* **Add New** Animation/Animation Blueprint
* [Create Animation Blueprint] Parent Class="VehicleAnimInstance"
* [Create Animation Blueprint] Target Skeleton="*MyVehicle*_Skeleton"
* **Select**
* Rename the asset "*MyVehicle*_Anim"

[image]

* Open the asset
* Select the "Anim Graph" Tab
* Right-clic menu [All actions for this Blueprint/Skeletal Control Nodes/Wheel Handler for Wheeled Vehicle]
* Link the "Wheel Handler" to the "Final Animation Pose"
* Right-clic menu [All actions for this Blueprint/Identity/Mesh Space Ref Pose]
* Link the "Mesh Space Ref Pose" to the "Wheel Handler"

[image]

* **Compile**
* **Save**
* Close the asset editor

## Add Vehicle Blueprint
* **Add New** Blueprint Class
* [Pick Parent Class/All Classes] Search="BaseVehiclePawn"
* **Select**
* Rename the asset "BP_*MyVehicle*"

[image]

* Open the asset
* [Pawn] AutoPossessPlayer="Player0"
* Select Mesh(Inherited)
* Mesh="*MyVehicle*_Mesh"
* [Animation] AnimClass="*MyVehicle*_Anim"
* Set Vehicle Bounds X, Y, Z
* Select VehicleMovement(Inherited)
* Associate each bone with a vehicle wheel:
    * [Vehicle Setup/Wheel Setups/*0-3*] WheelClass="*MyVehicle*_Wheel"
    * [Vehicle Setup/Wheel Setups/0] Bone Name="Wheel_Front_Left"
    * [Vehicle Setup/Wheel Setups/0] Bone Name="Wheel_Front_Right"
    * [Vehicle Setup/Wheel Setups/0] Bone Name="Wheel_Rear_Left"
    * [Vehicle Setup/Wheel Setups/0] Bone Name="Wheel_Rear_Right"
* Disable steering for rear wheel if required

[image]

* **Compile**
* **Save**
* Close the asset editor

## Add Vehicle Camera (optional)
* **Add Component**
* Search="SpringArm"
* **Add Component**
* Search="Camera"

[image]

* Select the "ViewPort" tab
* [Transform/Location] X=-500, Z=500
* [Transform/Rotation] Y=-25
* Select the "Event Graph" tab
* Right-clic menu [All actions for this Blueprint/Input/Axis Event/CameraRight]
* Right-clic menu [All actions for this Blueprint/Input/Axis Event/CameraUp]
* Dra&Drop "SpringArm" in the "ViewPort"
* Start a link...
* Select [Utilities/Transformation/AddRelativeRotation]
* Right-clic on "AddRelativeRotation/DeltaRotation" [Pin Actions/Split Struct Pin]
* Link "InputAxis CameraRight" to "AddRelativeRotation" (Delta Rotation Z)
* Link "InputAxis CameraUp" to "AddRelativeRotation" (Delta Rotation Y)

[image]

* **Compile**
* **Save**
* Close the asset editor

gsmario, 2020-06-30:
Hi! First let me thank you for this awesome tutorial you've created, its a very good job.
I'll try this workflow later, some parts are different from our workflow, but so far, it seems very complete and accurate. 
Thanks again!

villimikko, 2020-07-02:
Thanks very much for the guide it helped a lot. I have still some problem with the boundbox, for some reason i cant resize it. .
[image]

gsmario, 2020-07-02:
hi @villimikko try to expand the context menu, and search for the transform options, if pressing the R key in the viewport doesn't work. 
[image]

villimikko, 2020-07-02:
Thanks for that! For some reason when i spawn vehicle, the wheels work but car wont move at all(even pygame windows show acceleration and the wheels are spinning). Otherwise i think it is working.

villimikko, 2020-07-06:
[image]
[image]
[image]

When i have "hull" box the car drops throught terrain, but when i dont have it it wont move?

What did I do wrong this time? 👎

gsmario, 2020-07-06:
Hi @villimikko 
Please, be careful when creating and placing the collision boxes, if you put the wheels collision inside the bodywork (the car's general collision box) you'll have problems. Try to recreate this with your car: 
[image]

villimikko, 2020-07-20:
Tried to do that still having thse kinds of physics bugs on Carla.
[image]

What could cause this?

gsmario, 2020-07-20:
Hello @villimikko I'm asuming you're using blender, please, try to rig your wheels using only "rotation" as option, instead of "rotation and scale" and test it again. Also, set the wheels as kinematic in the physic asset.
I hope this helps.
[image]

chandeep11, 2020-07-21:
Hello @villimikko, check [this](https://www.youtube.com/watch?v=w70w1o3nddg) link for detailed steps on adding a car in UE4 from Blender.

villimikko, 2020-08-01:
I used this Blender addon to autorig my model  and after that i didint have any problems in carla.

Thanks everyone for helping me out!

https://continuebreak.com/creations/ue4-vehicle-rigging-addon-blender/

closed 2020-08-01 by villimikko

gaohongfein, 2020-10-29:
Hi villimikko,
I try to add a new vehicle in carka according with your tutorial. Now I can conrol the vehicle in manual control. The wheel can rotation and steering. But the vehicle can not move. Do you know how to solve this problem?
[image]

gaohongfein, 2020-10-29:
@villimikko

gaohongfein, 2020-10-29:
[image]
The vehicle is above the ground!

eslaaam3, 2022-08-18:
Hello,
I'm having a similar issue and I have posted an [issue](https://github.com/carla-simulator/carla/issues/5674) #5674
Can anyone help please?
Thanks!
@gsmario @maxime-derache

gokcerSnmzck, 2026-04-25:
Hello, I am experiencing the same issue regarding the non moving vehicle. I have opened a new thread with the screenshots at this post #9695 . Can you check it please ? I would really appreciate your help @germanros1987 @gsmario @chandeep11 @maxime-derache @villimikko