carla-simulator/carla #2926: Version Mismatch detected with 0.9.9 Windows compiled version

JeanPhilippeMonteuuis, 2020-06-07:
Unable to launch examples when downloading CARLA 0.9.9 Windows compiled version zip from releases page: https://carla-releases.s3.eu-west-3.amazonaws.com/Windows/CARLA_0.9.9.zip
How can I solve this version mismatch problem please ?

$ python .\spawn_npc.py
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version     = 0.9.7-2-g73f91e0b-dirty
WARNING: Simulator API version  = 0.9.9

Gunn3RR1, 2020-06-11:
for me at least, even though it says version mismatch, it still works

germanros1987, 2020-06-26:
@JeanPhilippeMonteuuis it seems that by accident you are using the egg of a previous version of CARLA (0.9.7). Make sure you get rid of it and use the right egg (0.9.9).

closed 2020-07-30 by germanros1987