carla-simulator/carla #2927: Coordinates of Bounding Box with respect to Ego Vehicle

dev015, 2020-06-08:
Hello,

The coordinates of the bounding box that we are currently getting in Carla are in the world frame, and I am trying to get these coordinates with respect to ego vehicle. I am trying to get a solution using computer vision basics and finding out the distance between the ego vehicle and other actors with a bounding box and then get the extent. 

Does anyone have any idea and help me with this.

Thanks !!

glopezdiest, 2020-06-12:
Hey @dev015 How are you getting the bounding box of the vehicle? I think that the [carla.BoundingBox](https://carla.readthedocs.io/en/latest/python_api/#carlaboundingbox) class has everything you are asking for. Or am I missunderstanding your question?

dev015, 2020-06-14:
Hey @glopezdiest, the coordinates of BB (bounding box) we get are in the world frame, but I am trying to get them with respect to ego vehicle instead of world frame.

glopezdiest, 2020-06-15:
Hmm... If by coordinates you mean vertices, you can use **_get_local_vertices(self)_**.

germanros1987, 2020-06-26:
@dev015 is your question answered?

dev015, 2020-06-26:
Hello @germanros1987, if I am not wrong if we need coordinates of an actor with respect to ego vehicle then we need to find the distance from ego to actor, is it right?

glopezdiest, 2020-06-29:
@dev015 Yes, sort of. You should get the world coordinates of the specific actor and then subtract the ones from the ego vehicle.

dev015, 2020-06-30:
Hello, @glopezdiest I have written a code for the same, do you mind if I can get your email id and discuss it with you?

glopezdiest, 2020-07-01:
I don't see the need to do so. You can send your code through here directly, if you want. However, if you don't, we can use Discord too, which is the main way we communicate with everyone. Here is my user tag *Guillermo Lopez#3589*

dev015, 2020-07-01:
@glopezdiest Thank you for your suggestions, this is solved.

closed 2020-07-01 by dev015

Banna74, 2021-02-05:
@dev015 Hey, me and my colleague are attempting to the same thing as you were trying back then. To calculate distance with respect to other vehicles on the road after applying the YOLO-v3 object detection but we are unable to, can you give us assistance or share the code reached? my mail is treka09@gmail.com if needed, thanks in advance.

dev015, 2021-02-05:
@Banna74 Hey, sure, can you please explain to me the exact problem you are facing.