carla-simulator/carla #2934: dynamically scalable vehicles

AlexScheid, 2020-06-09:
Hello CARLA community,

first of all thank's for such a great simulator.
This is not a real issue, more a suggestion to an easy to implement feature with a small question at the end.

I try to play a dataset in my CARLA simulation and therefore I also use some of the available features of CARLA like the OpenDRIVE standalone mode. While implementing the clientside, there occurs the problem that CARLA does actually not provide scalable vehicles. My dataset consists of many different cars and trucks and the difference between the already existing CARLA vehicles and some of my desired vehicles is to great to neglect this.

To solve this problem I created two very simple vehicles (you can see below) like described [hier](https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/). Because the simulation is intended for decision making, these simple vehicles are good enough.
Next I added the attributes "extent_x" and "extent_y" (*) to the vehicle parameters located at /<carla_repo>/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ VehicleParameters.cpp
[image]
To get access to these attributes and make them available for my vehicles I added the following lines to /<carla_repo>/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorBlueprintFunctionLibrary.cpp
[image]
Then I used _make clean_ and _make launch_.
To use this attributes only if one of my scalable vehicles is spawned I expand the "Spawn Actor" function of the "VehicleFactory" class in UE4 editor like shown in the picture below.
[image]
Then I added the function "Set Extent" (already used above) to the "BaseVehiclePawn" class to calculate the scaling factors and scale the actor.
[image]

This works quite well for spawning scaled vehicles during my simulation. I shared this here because I thought, that this is maybe a small feature needed by some more people.

One unsolved problem occurs in this process, marked with the star (*). I can only scale the x and y direction correctly. When I try to scale the actor in the z direction, the bounding box and the mesh of my actor got an offset towards each other. I think that is because the centers of scaling differ in z-direction like shown below.
[image]
I tried to compensate this by shifting the bounding box dynamically responding to a z-scaling factor in the new function "Set Extent", but I cannot get access to its object. Maybe somebody knows a solution therefore and a scalable object can be taken into account for a future version.

hbanzhaf, 2020-06-17:
Great new feature to accurately replay real-world data in CARLA. What's the CARLA team's (@glopezdiest, @sergi-e, @joel-mb) opinion on that? Would it make sense to turn this into a PR?

germanros1987, 2020-06-26:
@bernatx what is your opinion on this?

AlexScheid, 2020-07-02:
Update:
After some more tries, I found a more simple solution for this feature. I used the changes in the C++ Code shown above, but removed all the changes in the UE4-Editor files. A change of the transform used to spawn a vehicle in the "Spawn Actor" function of the "VehicleFactory" produces the same result. The part I added is shown below. It is like a summary of the old version combined with the integration in the transform.
[image]

Additionally I found one problem by scaling a vehicle in this way and I can give a possible solution for that. If you move a vehicle with the command "set_transform()" the CARLA-side implementation is using a location and a rotation while the UE4-side equivalent function is using also a scaling vector. This destroys the adapted scaling. To remove this problem, I had to change the C++ code of /<carla_repo>/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Server/CarlaServer.cpp at the function "BIND_SYNC(set_actor_transform)" like shown below.
[image]

AlexScheid, 2020-07-09:
Update:
After some tests there were still some points that had to be improved. For example the collision bodies of UE4 did not match with the optical mesh ("pxvis collision" in UE4 cmd) and the x-direction was not scalable.
These problems are solved with the this update and the scalable vehicle can be used as expected.
Changes:
As described in the first two images of this issue, I added the "extent_z" attribute. 
Then I changed the  "Spawn Actor" function of the "VehicleFactory" again. The final result looks like this:
[image]
The detailed changes are below.
[image]
[image]
[image]
[image]

Used float variables "scalable_vehicle_x", "scalable_vehicle_y" and "scalable_vehicle_z" are set to the unscaled dimensions of my model. The bool variable "scalable" distinguishes the scalable vehicle from the other vehicles. The vector variable "scaling vector" is used for clarity in the function.
Because of some problems with the wheel physics, I have to add some lines of code in the BeginPlay-function in the (<carla_repos>/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/) CarlaWheeledVehicle.cpp file
[image]

My last problem was the bounding box of the vehicle, which is referenced in some of the CARLA-functions. Its center is not shifted if a scaling was performed. Therefore I added a function to CarlaWheeledVehicle.cpp and CarlaWheeledVehicle.h to perform such a relative shift.
[image]

KunanonxGithub, 2020-09-16:
@AlexScheid So, if we want to change some parameters or change our cars to be bigger and be more realistic, we need to use C++ to adjust PhyX Engine in UE4. Am I right?

AlexScheid, 2020-09-16:
Hi @KunanonxGithub, as far as I remember you do not have to use C++ for a static change of the dimension or for the physical parameters at all. The issue should address a dynamic scalable vehicle, which means that you can use the same model to spawn vehicles with different dimensions (parameters are also conceivable) in one simulation. I had to change some C++ files of CARLA to get access to the newly created ability.
If you want to change the parameters once before you start a simulation (I called that static) and these parameters are fixed for all the vehicles of this type in your simulation, than you do not need any C++. All these things can be done in UE4. I can recommend you to look at the documentation [here](https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/). During this process you can change some physical parameters in the menus of UE4. (not done in this documentation) You do not have to use a new vehicle, you can get access to the already existing vehicles of CARLA and modify them.
Hope this will help you

closed 2020-12-25 by stale[bot]