carla-simulator/carla #2935: Error: computer automatically shutdown by running CarlaUE4.sh on Ubuntu 20.04

happengor, 2020-06-09:
I followed the build instruction by https://carla.readthedocs.io/en/latest/start_quickstart on Ubuntu 20.04.

After Installation i went into folder /opt/carla-simulator/bin and tried to run CarlaUE4.sh. It requires also entering password of sudo. 
As the scene showed up, the computer was turned off automatically. Could anybody explain this issue? Is the carlar 0.99 with deb Installation not compatible with ubuntu 20.04?

germanros1987, 2020-06-26:
@joel-mb could you try to launch the deb in a 20.04 distribution?

joel-mb, 2020-06-28:
Hi @happengor ! I'll try to launch the deb package in Ubuntu 20.04 during this week to see if there is some kind of incompatibility. Regarding the issue entering the password, there is a PR fixing this (https://github.com/carla-simulator/carla/pull/2945). As soon as it gets merged I'll update the package.

happengor, 2020-06-30:
Thanks, i have solved my problem with a bigger psu.

happengor, 2020-06-30:
Thanks, guys, i have solved my problem with a bigger psu. Now i can launch the deb in 20.04.

joel-mb, 2020-06-30:
Perfect! Thanks for the information!

closed 2020-06-30 by joel-mb