carla-simulator/carla #2938: Collision detection with specific vehicle in CARLA 0.8

noahcao, 2020-06-10:
Hi, thanks very much for the great simulator.

I'm using CARLA 0.8.4 on Ubuntu/Windows. I'm doing some planning work and would like to know which vehicle the ego-vehicle collides with when a collision happens. I checked the document but found no way to implement it. Is it possible to have such a function? Or, if not, is there some trick to approximate such a function?

Armandomiguel, 2020-06-17:
Hello noahcao,

You should take a look to the Collision Sensor (https://carla.readthedocs.io/en/latest/ref_sensors/#collision-detector).
When attached to an actor, a predefined method is called whenever a collision involving this actor occurs; and the output parameters include the ID of both actors involved.

PS. I really recommend you using a more recent version of CARLA (0.9.7 or newer) and checking the documentation for sensors in order to know how to set them and declare their callback functions.

Best regards,

noahcao, 2020-06-24:
Hi Armandomiguel,

Thanks so much for the kind help!

yes, I realized there is some solution on CARLA 0.9. But due to some legacy codebase issue and to have fair comparison with some other works which are based on stable CARLA version (0.8.4), I'm still expecting a solution on CARLA 0.8.4.

But you're right. I'm currently migrating my code to CARLA 0.9.

germanros1987, 2020-06-26:
The collision sensor was available in 0.8.4, wasn't it, @marcgpuig ?

marcgpuig, 2020-06-26:
Hi @noahcao 
Sorry, but the collision sensor was added in 0.9.1.
I highly recommend porting the code to a newer version.
Cheers!

closed 2020-09-05 by stale[bot]

TzabarDolev, 2023-09-18:
Hi,
I just wrote a short blog post about collision detectors on the Carla simulator research blog - https://carlasimblog.wordpress.com/2023/09/18/collision-detection-systems-in-cars-enhancing-safety-and-realism-in-carla-simulator/, and added a code example for it.