carla-simulator/carla #2939: IMUMeasurement gyroscope values in world-frame instead of body-frame

GabiShahmayster, 2020-06-10:
Hello,

First of all, thank you very much for this great simulator, I've been using it extensively for the last few days and so it's been a great experience.
I'm currently focused on retrieving IMU data and feeding it into a GNSS/INS fusion algorithm, so the first thing is to ensure that IMU data is properly received and fed to the INS.
I've started using version 0.9.9.1, which fixed a bug related to the projection of gravity to vehicle body axes. 
I'm using Ubuntu 18.04.04 LTS and running the pre-built binaries download from the Carla repo.

I'm encountering an issue related to the gyroscopes measurements. They seem to be output in the world frame, instead of the vehicle body frame. 
The accelerometers look ok. 
This issue can be easily reproduced by driving a vehicle in a straight line (I'm using VehiclePIDController to drive the vehicle for a few tens of meters). 
During the drive, the vehicle pitch fluctuates. This should be reflected in a corresponding Y axis rotation-rate, in the vehicle-frame. The vehicle experiences acceleration, which should be reflected mostly in X-axis acceleration, in the vehicle-frame. 
We can now drive the vehicle, in a straight line, in two different orientations:
1. Vehicle is east-bound (UE world yaw = 0[deg]). IMU-frame should be aligned with UE-frame.
2. Vehicle is south-bound (UE world yaw = 90[deg]). IMU-frame X should be UE-frame Y, IMU-frame Y is UE-frame (-X). 

In both scenarios, we would expect similar vehicle pitch profile and body-frame IMU measurements.
The actual results show a similar vehicle pitch profile and accurately reported body-frame IMU accelerometers values.
However it seems as though the gyroscopes report world-frame rotation-rates... 
We can see the issue in the following figures:
East-bound scenario has similar world/imu accelerations/rotation-rates, as expected.
South-bound scenario has different world/imu accelerations, as expected BUT similar rotation-rates, which shouldn't be the case, because the vehicle is rotated w.r.t the world.

I've retrieved the world-frame accelerations and rotation-rates using Actor's get_acceleration and get_angular_velocity() methods.

p.s I'm sorry if this is too much information for a very simple issue to describe...

Thank you very much!
Gabi Shahmayster

[image]
[image]
[image]
[image]
[image]
[image]

GabiShahmayster, 2020-06-22:
Anyone from the Carla team? @ItsTimmy could you please take a look? 
Maybe I'm missing something fundamental regarding Carla or there's any non-standard IMU modelling going on...

ItsTimmy, 2020-06-23:
I'm not actually part of the Carla team, but I can still take a look. This issue might actually explain some other problems I've been having with Carla...

GabiShahmayster, 2020-06-23:
> I'm not actually part of the Carla team, but I can still take a look. But I don't have a lot of free time, so I probably will not get to it soon.
> 
> So if anyone else comes across this issue, don't wait for me to fix it!

@ItsTimmy Thank you very much. Unfortunately, I am not familiar with the official Carla team or the most effective ways of reporting such an issue... If you think that I should contact someone from the team, via Discord chat perhaps, or try to fix the issue and submit a PR, please tell me so...

germanros1987, 2020-06-25:
@marcgpuig and @DSantosO could you take a look? I know it's a busy time... but this was supposed to be solved. It's not the first time we find similar problems.

ItsTimmy, 2020-06-26:
I have a fix in #2998 . Still needs some testing though.

DSantosO, 2020-06-26:
Thank you @GabiShahmayster for bringing the issue and @ItsTimmy for the fix. We will test it and hopefully merge it as soon as possible.

closed 2020-07-02 by marcgpuig

wu0023, 2022-03-29:
Hi， I'm wondering whether this bug is fixed in the old version, (e.g., 0.9.9.4)?

GabiShahmayster, 2022-03-29:
Hi @wu0023,
It was fixed in version 0.9.10 (https://github.com/carla-simulator/carla/releases/tag/0.9.10/)

PR #2998 (2020-07-02): Fixed gyro to report angular velocity in local sensor frame
#### Description

The gyroscope in the IMU Sensor reported angular velocity in world frame. Real IMUs would report it in the local sensor frame. This PR fixes that.

Fixes #2939. 

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 2.7 and 3.6
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbacks

None known

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/2998)