carla-simulator/carla #2942: Landmark  is not linked to any traffic light

hepeng187, 2020-06-11:
```
        temp_map = self.world.get_map()
        for landmark in temp_map.get_all_landmarks_of_type('1000001'):
            if landmark.id != '':
                traffic_light = self.world.get_traffic_light(landmark)
                if traffic_light is not None:
                    self._tls[landmark.id] = traffic_light 
                else:
                    logging.warning('Landmark %s is not linked to any traffic light', landmark.id)
```
the landmark exists, but is not linked to traffic lights.  This bothers me, what am I doing wrong?

closed 2020-09-05 by stale[bot]

kargarisaac, 2022-02-16:
Does this help? (https://github.com/carla-simulator/carla/blob/master/Co-Simulation/Sumo/util/netconvert_carla.py#L415)
```python
if landmark.name == '':
      # This is a workaround to avoid adding traffic lights without controllers.
      logging.warning('Landmark %s has not a valid name.', landmark.name)
      continue
```

lpwwpl, 2022-08-26:
> ```
>         temp_map = self.world.get_map()
>         for landmark in temp_map.get_all_landmarks_of_type('1000001'):
>             if landmark.id != '':
>                 traffic_light = [...]

i have the same problem, do u have solved it?

Henlenteo, 2024-09-11:
Hello, when I tried to import the custom map built by Roadrunner into Carla, I also found through debugging that the landmark cannot be associated with traffic light. I guess this is caused by an incorrect use of signals in the OpenDrive file. If you could help me, I would greatly appreciate it!