carla-simulator/carla #2943: Segmentation Fault at world.getMap()

akramer123, 2020-06-11:
I´m compiling CarlaLib myself following the instructions for the c++ client on this page:
https://github.com/carla-simulator/carla/tree/master/Examples/CppClient

But after including the lib to my program and when i´m calling world.getMap(),  i´m getting a segmentation fault :

void std::vector<carla::road::Lane*, std::allocator<carla::road::Lane*> >::emplace_back<carla::road::Lane*>(carla::road::Lane*&&) at /usr/include/c++/7/bits/vector.tcc:98
0x00007f2423227040 in /home/aadc/AADC/_install/linux64/bin/Simulation_Input.adtfplugin: carla::road::MapBuilder::CreatePointersBetweenRoadSegments() at /home/aadc/AADC/build-debug/src/aadcUser/simulation/input/CarlaPrj-prefix/src/CarlaPrj/Build/../LibCarla/source/carla/road/MapBuilder.cpp:705
0x00007f242322dc62 in /home/aadc/AADC/_install/linux64/bin/Simulation_Input.adtfplugin: carla::road::MapBuilder::Build() at /usr/include/c++/7/bits/hashtable.h:492
0x00007f24231fcc82 in /home/aadc/AADC/_install/linux64/bin/Simulation_Input.adtfplugin: carla::opendrive::OpenDriveParser::Load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /usr/include/c++/7/bits/stl_vector.h:434

Thx in advance!

germanros1987, 2020-06-26:
@akramer123 could you provide more information about your system, version of carla, etc?

kwetril, 2020-07-01:
for me, the error reproduces with carla `0.9.9` and `master`.
I tried ubuntu 16.04 and 18.04

kwetril, 2020-07-01:
here is a similar issue: https://github.com/carla-simulator/carla/issues/2079

akramer123, 2020-07-01:
Yes, i´m also working with carla 0.9.9 and master and running it on Ubuntu 18.04 LTS.
@germanros1987

bernatx, 2020-07-01:
Hi, that problem was fixed in patch version 0.9.9.2
It was related to a workaround for an Unreal bug, that is also fixed with an Unreal patch on 0.9.9.3
I hope it works for you.

kwetril, 2020-07-01:
just tried with `0.9.9.3` version, the issue still reproduces.
I have `next_lane == nullptr` here https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/road/MapBuilder.cpp#L708
It looks like there is a problem with parsing of .xodr map file.

bernatx, 2020-07-01:
Are you trying to load a CARLA map or a custom one?
It seems that the .xodr is malformed.

kwetril, 2020-07-01:
I just run `make launch`, UE Editor opens Carla's map "Town03". I play this map, connect to Carla from client app and try to get map with `world.getMap()`. I also tried other default maps but got the same error

kwetril, 2020-07-01:
xodr files doesn't seem to be corrupted, as the test `libcarla_test_client_release` parses them successfully

akramer123, 2020-07-02:
I also tested it with version 0.9.9.4 and got same segmentation fault.

kwetril, 2020-07-16:
@bernatx any ideas what can we do to help investigate the problem?

kwetril, 2020-07-16:
when I build my app with `carla_client_debug.a`, this assertion fails:
```
(gdb) bt
#0  0x00007fffeee0ef47 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007fffeee108b1 in __GI_abort () at abort.c:79
#2  0x00007fffeee0042a in __assert_fail_base
    (fmt=0x7fffeef87a38 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=assertion@entry=0x7ffff78d6ee7 "next_lane != nullptr", file=file@entry=0x7ffff78fbd46 "../../LibCarla/source/carla/road/MapBuilder.cpp", line=line@entry=707, function=function@entry=0x7ffff78fcd73 "void carla::road::MapBuilder::CreatePointersBetweenRoadSegments()") at assert.c:92
#3  0x00007fffeee004a2 in __GI___assert_fail
    (assertion=0x7ffff78d6ee7 "next_lane != nullptr", file=0x7ffff78fbd46 "../../LibCarla/source/carla/road/MapBuilder.cpp", line=707, function=0x7ffff78fcd73 "void carla::road::MapBuilder::CreatePointersBetweenRoadSegments()") at assert.c:101
#4  0x00007ffff770cf04 in carla::road::MapBuilder::CreatePointersBetweenRoadSegments() (this=0x7fffffffb828) at ../../LibCarla/source/carla/road/MapBuilder.cpp:707
#5  0x00007ffff770c9e3 in carla::road::MapBuilder::Build() (this=0x7fffffffb828) at ../../LibCarla/source/carla/road/MapBuilder.cpp:38
#6  0x00007ffff76ad423 in carla::opendrive::OpenDriveParser::Load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)Python Exception <class 'gdb.error'> There is no member named _M_dataplus.: 
 (opendrive=)
    at ../../LibCarla/source/carla/opendrive/OpenDriveParser.cpp:49
--Type <RET> for more, q to quit, c to continue without paging--
#7  0x00007ffff768e3c0 in carla::client::MakeMap(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)Python Exception <class 'gdb.error'> There is no member named _M_dataplus.: 
 (opendrive_contents=) at ../../LibCarla/source/carla/client/Map.cpp:29
#8  0x00007ffff768e246 in carla::client::Map::Map(carla::rpc::MapInfo) (this=0x55557c8467e0, description=...) at ../../LibCarla/source/carla/client/Map.cpp:38
#9  0x00007ffff75cd197 in boost::make_shared<carla::client::Map, carla::rpc::MapInfo>(carla::rpc::MapInfo&&) (args=...) at ../boost-1.72.0-c8-install/include/boost/smart_ptr/make_shared_object.hpp:256
#10 0x00007ffff75c646f in carla::MakeShared<carla::client::Map, carla::rpc::MapInfo>(carla::rpc::MapInfo&&) (args=...) at ../../LibCarla/cmake/../source/carla/Memory.h:30
#11 0x00007ffff75c6405 in carla::client::detail::Simulator::GetCurrentMap() (this=0x55557dd1ca80) at ../../LibCarla/source/carla/client/detail/Simulator.cpp:129
#12 0x00007ffff74d1400 in carla::client::World::GetMap() const (this=0x7fffffffc2a0) at ../../LibCarla/source/carla/client/World.cpp:22
```

bernatx, 2020-07-16:
Is that happening with Town03?

kwetril, 2020-07-16:
With other maps too, but when the same code is executed with python client - method `carla::client::MakeMap()` works having the same input. Maybe there is a conflict with some libs from my app, for example I have boost 1.68 and carla uses 1.72.
Will try to reproduce with CppClient from Examples

kwetril, 2020-07-16:
yep, it's something related to dependencies in my app, with [CppClient](https://github.com/carla-simulator/carla/tree/master/Examples/CppClient) the error doesn't reproduce for me

bernatx, 2020-07-16:
Probably the boost version, because we use 1.72 as you say, but also we patch two files of boost to work with openDRIVE, so it seems both things are related to the problem. Try with the proper version of boost (CARLA will patch automatically if you use the **make setup** command).

kwetril, 2020-07-16:
It's even more interesting now, the problem is related to Qt, I created a simple main() function and a single line causes crash:
```
int main(int argc, char *argv[]) {
  QApplication a(argc, argv);
...
```
without `QApplication` world.getMap() works, when uncommented - segmentation fault.
attempt to use boost from carla doesn't have any impact

kwetril, 2020-07-17:
tried different qt versions: 5.9.9, 5.12.9, 5.15.0 - everything crashes

kwetril, 2020-07-17:
Finally, I identified the problem, it's related to the system locale, looks like Qt affects it somehow.
The default separator for floating-point numbers changes from `.` to `,`, it breaks xml parsing and only integer part of road lengths is used.
For me, the problem was fixed after setting locale manually:
```
std::setlocale(LC_ALL, "en_US.UTF-8");
```
@akramer123 , could you please check, if it helps you too?

akramer123, 2020-07-18:
@kwetril 
Thx! That fixed our problem!

hh0rva1h, 2020-08-17:
Sounds similar to an issue we have been experiencing from the python side: https://github.com/carla-simulator/carla/issues/3194

KasraMokhtari2112, 2020-08-27:
> Finally, I identified the problem, it's related to the system locale, looks like Qt affects it somehow.
> The default separator for floating-point numbers changes from `.` to `,`, it breaks xml [...]

Can you please explain more where these lines of code should be modified? Thanks a lot.

closed 2021-04-27 by stale[bot]