carla-simulator/carla #2949: Taking camera reading as input in OpenCV

AmirHarvey, 2020-06-12:
Hello there,
I added two RGB cameras and they work fine, but instead of displaying them i'm trying to take their reading as OpenCV input to apply object detection on them.
I tried cv2.VideoCapture() but it can't take numpy array as input.

Anyone can help?
Thanks in advance.

I am using:
carla 0.9.8
python 3.7
os: windows

GabiShahmayster, 2020-06-13:
Hi Amir,
You don't need a VideoCapture object for that, since the video frames are provided directly by Carla. 
You would use a VideoCapture object to open a webcam video stream, video file or images folder, and it would provide you with a sequence of frames, which Carla already does. 
Any subsequent image processing or computer vision algorithms, can be applied to this sequence of numpy arrays. You would then use OpenCV's imshow() and waitKey() methods to display the results, at a proper frame-rate.
Hope this helps in any way...
Gabi

AmirHarvey, 2020-06-14:
Thank you so much Gabi
This helped yes, now i'm not trying with the VideoCapture anymore but i'm still trying to figure out how to apply YOLOv3 on a numpy array because all the methods i could find only use images, videos or webcam as input.

germanros1987, 2020-06-26:
@AmirHarvey this is not the right place for this question. Please, try the [carla forum](forum.carla.org). Let's cooperate so that Github remains for bug reports, features requests, and questions **very** related to the mechanics of the CARLA API

closed 2020-06-26 by germanros1987