carla-simulator/carla #2953: Could not install carla-hdmaps

atinfinity, 2020-06-15:
I read <https://drive.google.com/file/d/1uO6nBaFirrllb08OeqGAMVLApQ6EbgAt/view?usp=sharing>.
And, I tried to install `carla-hdmaps` on Ubuntu 18.04.

```shell
$ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys BA0F9B7406F60E23
$ sudo add-apt-repository "deb [arch=amd64 trusted=yes] http://dist.carla.org/carla-hdmaps/ bionic main"
$ sudo apt-get update
$ sudo apt install carla-hdmaps
```

But, I could not install `carla-hdmaps`.

```shell
$ sudo apt install carla-hdmaps
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 carla-hdmaps : Depends: carla but it is not installable
E: Unable to correct problems, you have held broken packages.
```

pietrocolombo, 2020-06-23:
i have the same problem

germanros1987, 2020-06-25:
@joel-mb could you please take a look at this?

joel-mb, 2020-07-02:
Hi @atinfinity @pietrocolombo !
Can you try again? I've removed the `carla` dependency for this package. Remember to run `sudo apt-get update` before.

pietrocolombo, 2020-07-02:
@joel-mb thanks it works

joel-mb, 2020-07-02:
@pietrocolombo thanks for the feedback!

closed 2020-07-02 by joel-mb

atinfinity, 2020-07-02:
@joel-mb Thanks. I have successfully installed `carla-hdmaps`.